CameraKit/camerakit-android #428: Camera can't focus and take a picture on Samsung S6

BrunoTrovoN26, 2018-10-01:
**Environment**

**Development:**
Android Studio 3.2
Build #AI-181.5540.7.32.5014246, built on September 17, 2018
JRE: 1.8.0_152-release-1136-b06 x86_64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
macOS 10.13.6

**Running Devices:**
- Samsung S6

**Steps to Reproduce:**
Open the camera using camera kit and the device mentioned above, and try to take a picture.

**Expected Behavior**
Should establish focus and execute the _Take Picure_ action.

**Actual Behavior**
When trying to take a picture, the camera in the mentioned device is not able to focus properly, and since it can't focus, it doesn't execute the _Take Picture_ action. Also, All focus modes were tried.

This does not happen for me on the following devices (i.e. they look perfect):
- Moto G5 (Android 7.0)
- Moto Z2 A5 (Android 8.0)

srinathvidapu, 2018-10-10:
same issue for redmi note 4 image callback not callling

lwlizhe, 2018-10-18:
same issue for huawei honor 8，cameraKitView.captureImage(CameraKitView.ImageCallback)  doesn't work

austinkettner, 2018-10-26:
Hi guys! Can you please see ticket #383 I have added the reported devices but was hoping you each could comment with Android OS and Device ID to make tracking these issues more precise.

Also @srinathvidapu can you comment and confirm in that ticket that the issue is focus related? You are specifically mentioning a callback not returning, so wanted to confirm. Thanks!

closed 2018-10-26 by austinkettner