CameraKit/camerakit-android #429: Fatal Exception: java.lang.RuntimeException setParameters failed Camera1.java line 295 com.wonderkiln.camerakit.Camera1.setZoom

jewom, 2018-10-04:
I receive this crash from Crashlytics :

Fatal Exception: java.lang.RuntimeException setParameters failed
Camera1.java line 295
com.wonderkiln.camerakit.Camera1.setZoom

With a TOSHIBA Tab AT10LE-A/AT15LE-A/AT10PE-A/AT15PE-A (Japan: AT703)

[com.belami.essential_issue_58_crash_5BB4D8AB018900016D0A187E3A8DFFEE_DNE_0_v2.txt](https://github.com/CameraKit/camerakit-android/files/2445825/com.belami.essential_issue_58_crash_5BB4D8AB018900016D0A187E3A8DFFEE_DNE_0_v2.txt)

emersoncloud, 2018-10-05:
Hi @jewom which version of CameraKit are you using? Also which version of Android was the device running that crashed? Thanks!

coderYiTaoZhang, 2018-10-22:
I'm using version 0.13.0 ,I have  met the same problem; The version of Android is 6.0.1 8.0.0 7.1.2 7.1.1 7.0;
Mabe you need the devices types.there are some   oppoR9s R9; SM-G930S  VIVO Y66 M6 Note

austinkettner, 2018-11-19:
Hey @coderYiTaoZhang we are not longer supporting 0.13.x releases, and are exclusively focused on v1.0.0 via #318 - please use that instead.

closed 2018-11-19 by austinkettner