CameraKit/camerakit-android #430: Camera1.java line 295 com.wonderkiln.camerakit.Camera1.setZoom  Fatal Exception: java.lang.RuntimeExceptionsetParameters failed

jewom, 2018-10-04:
Camera1.java line 295
com.wonderkiln.camerakit.Camera1.setZoom

Fatal Exception: java.lang.RuntimeExceptionsetParameters failed Raw Text
--
  | android.hardware.Camera.native_setParameters (Camera.java)
  | android.hardware.Camera.setParameters (Camera.java:1497)
  | com.wonderkiln.camerakit.Camera1.setZoom (Camera1.java:295)
  | com.wonderkiln.camerakit.Camera1.adjustCameraParameters (Camera1.java:845)
  | com.wonderkiln.camerakit.Camera1.adjustCameraParameters (Camera1.java:779)
  | com.wonderkiln.camerakit.Camera1.openCamera (Camera1.java:671)
  | com.wonderkiln.camerakit.Camera1.start (Camera1.java:124)
  | com.wonderkiln.camerakit.Camera1.setFacing (Camera1.java:192)
  | com.wonderkiln.camerakit.CameraView$3.run (CameraView.java:339)
  | android.os.Handler.handleCallback (Handler.java:730)
  | android.os.HandlerThread.run (HandlerThread.java:61)

[com.belami.essential_issue_58_crash_5BB4D8AB018900016D0A187E3A8DFFEE_DNE_0_v2 (2).txt](https://github.com/CameraKit/camerakit-android/files/2445834/com.belami.essential_issue_58_crash_5BB4D8AB018900016D0A187E3A8DFFEE_DNE_0_v2.2.txt)

closed 2018-10-11 by jewom

manujjagga421, 2019-03-31:
@jewom any update on this? I am also facing the same issue.

jewom, 2019-04-01:
No :(

Abhay7591, 2020-08-20:
@jewom any update on this? I am also facing the same issue. I am using v0.13.4