CameraKit/camerakit-android #431: D8 issues

devonEdison, 2018-10-05:
### Is this a bug report?

no

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

yes

### Environment
mac, android studio 3.2

### Steps to Reproduce

(Write your steps here:)
https://issuetracker.google.com/issues/77842465
since the D8 has been used in android studio 3.2
so the gradle build might have some issues when installing app. (takes so much time.)

please upgrade the gradle to the latest version thanks

### Expected Behavior

should install app immediately
### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

devonEdison, 2018-10-05:
if I uses this library , it will cause  slow install on a few phones. (like LG G4)

devonEdison, 2018-10-05:
please help

austinkettner, 2018-10-05:
Hi @devonEdison can you elaborate on what you mean by 'Slow install'? Do you have specific metrics you can list here to demonstrate the difference?