CameraKit/camerakit-android #437: 3.9 beta -- camerakitview.captureimage/capturevideo

tt-aali, 2018-10-13:
Do those callbacks still exist? Or are we capturing snapshots and videos in a different way?

emersoncloud, 2018-10-15:
Hi @tt-aali we will update the documentation to reflect the latest changes in 1.0.0 when we release beta4. The `camerakitview.captureImage` method still exists and expects a `CameraKitView.ImageCallback()`. See the example below:

```
cameraKitView.captureImage(new CameraKitView.ImageCallback() {
                @Override
                public void onImage(CameraKitView cameraKitView, final byte[] photo) {
                     // do something with the photo
                }
}
```

Currently `captureVideo` is empty but will exist in 1.0.0-beta5!

closed 2018-10-15 by emersoncloud

fahime-ghasemi, 2018-11-03:
How can we take video when we are using 1.0.0-beta3.9?