CameraKit/camerakit-android #443: Camera kit crashes when flash set to auto

sanoopsandy, 2018-10-22:
### Is this a bug report?

Yes

Was trying to create a dummy project with CameraVeiw. After following the Readme section the app was crashing. Upon going through the stack-trace I saw that there was some issue with the flash section. 

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Kotlin

### Steps to Reproduce

1. Just add the camera kit to a new project and follow the steps given in Readme. 
2. Set the flash to auto in the xml `app:camera_flash="auto"`
3. When you run it the app will crash with following error

`java.lang.IllegalArgumentException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkParameterIsNotNull, parameter <set-?>`

### Expected Behavior
Was expecting the camera to open.

### Actual Behavior
It just crashed with the above error.

KlaartjeD, 2018-10-23:
It's the same with FLASH_TORCH

tolstovdmit, 2018-10-23:
Hi. I see at the website docs that the property name is actually ckFlash.
http://docs.camerakit.website/#/?id=ckflash

At the demo app too:
https://github.com/CameraKit/camerakit-android/blob/master/demo/src/main/res/layout/activity_main.xml

I think it's maybe more of the README.MD example changing.

KlaartjeD, 2018-10-23:
The property used to be ckFlash before 1.0, it's camera_flash now
app:camera_flash="on" and app:camera_flash="off" are both working.

tolstovdmit, 2018-10-23:
I see this paramter (camera_flash) is used to Came**Kit**View and the issue says CameraView. Maybe that's the problem?

tolstovdmit, 2018-10-23:
Can you provide a stacktrace of an error?

emersoncloud, 2018-10-23:
@KlaartjeD @tolstovdmit I apologize for the confusion. The error that @sanoopsandy originally posted is in CameraKit v1.0.0-beta3.9. The links to the documentation and demo application are in version 0.13.2: 

> http://docs.camerakit.website/#/?id=ckflash
> https://github.com/CameraKit/camerakit-android/blob/master/demo/src/main/res/layout/activity_main.xml

I'll be pushing a fix for the flash today and updating the master branch with the latest beta code shortly thereafter. Let me know if you have any more questions.

huangyz0918, 2018-10-28:
I was confused by your document, can it be updated as the latest version in the website? I can see there is still at version: 0.12.0, which is out of date :(

austinkettner, 2018-10-29:
Updates are in the works @huangyz0918 we know this is an important step to clarify many things!

closed 2018-10-29 by austinkettner