CameraKit/camerakit-android #444: [Beta3.9] Camera locks up upon capture

austinkettner, 2018-10-25:
## Official Maintainer Ticket

Issue Reported: Camera upon capture locks up, but does not crash.

| Device Name | Device ID | OS | CameraKit QA'd  | Expensify QA'd  |
|---|---|---|---|---|
|  Sony Xperia Z5 | E5823 | 7.1.1  |  |   | 

#### Rule to Close
- Only check off the devices that are tested
- Non-directly testable devices by Official CameraKit Team will be QA'd externally

GuanacoDevs, 2018-11-20:
Have the same issue with a XZ5P

UmarBhutta, 2018-11-28:
Same Happening on Sony Xperia Premium. 
But also there are some Crashes that i can see on my crashlytics are for Sony Devices, i.e 
Sony H8266 (Xperia XZ2) most effected device.
Sony F8332 (Xperia XZ).

i am pasting the log also.
Fatal Exception: java.lang.RuntimeException
       at com.camerakit.api.camera2.Camera2.open(SourceFile:48)
       at com.camerakit.api.ManagedCameraApi.open(SourceFile:12)
       at com.camerakit.CameraPreview.openCamera(SourceFile:235)
       at com.camerakit.CameraPreview$start$1$1.doResume(SourceFile:119)
       at kotlin.coroutines.experimental.jvm.internal.CoroutineImpl.resume(SourceFile:42)
       at kotlinx.coroutines.experimental.DispatchedTask$DefaultImpls.run(SourceFile:150)
       at kotlinx.coroutines.experimental.DispatchedContinuation.run(SourceFile:14)
       at kotlinx.coroutines.experimental.EventLoopBase.processNextEvent(SourceFile:136)
       at kotlinx.coroutines.experimental.BlockingCoroutine.joinBlocking(SourceFile:133)
       at kotlinx.coroutines.experimental.BuildersKt__BuildersKt.runBlocking(SourceFile:52)
       at kotlinx.coroutines.experimental.BuildersKt.runBlocking(Unknown Source:1)
       at kotlinx.coroutines.experimental.BuildersKt__BuildersKt.runBlocking$default(SourceFile:41)
       at kotlinx.coroutines.experimental.BuildersKt.runBlocking$default(Unknown Source:1)
       at com.camerakit.CameraPreview$start$1.doResume(SourceFile:116)
       at kotlin.coroutines.experimental.jvm.internal.CoroutineImpl.resume(SourceFile:42)
       at kotlinx.coroutines.experimental.DispatchedTask$DefaultImpls.run(SourceFile:150)
       at kotlinx.coroutines.experimental.DispatchedContinuation.run(SourceFile:14)
       at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
       at java.util.concurrent.FutureTask.run(FutureTask.java:266)
       at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run(Thread.java:764)

vssh, 2019-04-29:
We have the same problem:

```
java.lang.RuntimeException
	at com.camerakit.api.camera2.Camera2.open(Camera2.java:48)
	at com.camerakit.api.ManagedCameraApi.open(ManagedCameraApi.java:12)
	at com.camerakit.CameraPreview.openCamera(CameraPreview.java:234)
	at com.camerakit.CameraPreview$start$1$1.invokeSuspend(CameraPreview.java:118)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl.java:32)
	at kotlinx.coroutines.DispatchedTask$DefaultImpls.run(DispatchedTask.java:235)
	at kotlinx.coroutines.DispatchedContinuation.run(DispatchedContinuation.java:81)
	at kotlinx.coroutines.EventLoopBase.processNextEvent(EventLoopBase.java:123)
	at kotlinx.coroutines.BlockingCoroutine.joinBlocking(BlockingCoroutine.java:69)
	at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking(BuildersKt__BuildersKt.java:45)
	at kotlinx.coroutines.BuildersKt.runBlocking(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.runBlocking$default(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.withContext(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.async(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.async$default(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking$default(BuildersKt__BuildersKt.java:35)
	at kotlinx.coroutines.BuildersKt.runBlocking(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.runBlocking$default(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.withContext(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.async(BuildersKt.java:1)
	at kotlinx.coroutines.BuildersKt.async$default(BuildersKt.java:1)
	at com.camerakit.CameraPreview$start$1.invokeSuspend(CameraPreview.java:115)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl.java:32)
	at kotlinx.coroutines.DispatchedTask$DefaultImpls.run(DispatchedTask.java:235)
	at kotlinx.coroutines.DispatchedContinuation.run(DispatchedContinuation.java:81)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:458)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:764)
```

It seems to affect only Sony devices. Affected devices we found:
Sony F5321
Sony H8314
Sony H9436
Sony G8141
Sony H8416
Sony G8441
Sony H8166
Sony E5823

putrasto, 2019-07-24:
I had the same problem. Here is the reported exception by crashlytics.

- Device Name : Samsung J6, Samsung Galaxy S6, Sony Xperio X compact
- OS : Android 8 & Android 9

```
Fatal Exception: java.lang.RuntimeException
       at com.camerakit.api.camera2.Camera2.open + 48(Camera2.kt:48)
       at com.camerakit.api.ManagedCameraApi.open + 12(ManagedCameraApi.kt:12)
       at com.camerakit.CameraPreview.openCamera + 235(CameraPreview.kt:235)
       at com.camerakit.CameraPreview$start$1$1.doResume + 119(CameraPreview.kt:119)
       at kotlin.coroutines.experimental.jvm.internal.CoroutineImpl.resume + 42(CoroutineImpl.kt:42)
       at kotlinx.coroutines.experimental.DispatchedTask$DefaultImpls.run + 150(Dispatched.kt:150)
       at kotlinx.coroutines.experimental.DispatchedContinuation.run + 14(Dispatched.kt:14)
       at kotlinx.coroutines.experimental.EventLoopBase.processNextEvent + 136(EventLoop.kt:136)
       at kotlinx.coroutines.experimental.BlockingCoroutine.joinBlocking + 133(Builders.kt:133)
       at kotlinx.coroutines.experimental.BuildersKt__BuildersKt.runBlocking + 52(Builders.kt:52)
       at kotlinx.coroutines.experimental.BuildersKt.runBlocking + 1(:1)
       at kotlinx.coroutines.experimental.BuildersKt__BuildersKt.runBlocking$default + 41(Builders.kt:41)
       at kotlinx.coroutines.experimental.BuildersKt.runBlocking$default + 1(:1)
       at com.camerakit.CameraPreview$start$1.doResume + 116(CameraPreview.kt:116)
       at kotlin.coroutines.experimental.jvm.internal.CoroutineImpl.resume + 42(CoroutineImpl.kt:42)
       at kotlinx.coroutines.experimental.DispatchedTask$DefaultImpls.run + 150(Dispatched.kt:150)
       at kotlinx.coroutines.experimental.DispatchedContinuation.run + 14(Dispatched.kt:14)
       at java.util.concurrent.Executors$RunnableAdapter.call + 459(Executors.java:459)
       at java.util.concurrent.FutureTask.run + 266(FutureTask.java:266)
       at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run + 301(ScheduledThreadPoolExecutor.java:301)
       at java.util.concurrent.ThreadPoolExecutor.runWorker + 1167(ThreadPoolExecutor.java:1167)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run + 641(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run + 764(Thread.java:764)
```