CameraKit/camerakit-android #446: pinch to zoom doesn't work

fahime-ghasemi, 2018-10-29:
I used the document if master branch for using camerakit in my app. According to the documentation, the library has pinch to zoom feature, but there isn't.

austinkettner, 2018-10-29:
Hi @fahime-ghasemi which version of CameraKit are you using?

fahime-ghasemi, 2018-10-29:
I added the folowing codes to my build.gradle

    implementation 'com.camerakit:camerakit:1.0.0-beta3.9'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.2.61'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:0.24.0'

austinkettner, 2018-10-29:
Pinch to Zoom is not currently in the v1.0.0 releases. We expect this to return with full documentation in Beta4. You can track this feature here: https://github.com/CameraKit/camerakit-android/issues/447