CameraKit/camerakit-android #447: [Feature] Pinch to Zoom

austinkettner, 2018-10-29:
### Official Maintainer Ticket

- [ ] Add Pinch to Zoom functionality
- [ ] Add Documentation of Feature

android-dataticket, 2019-03-13:
any work on this?

Corical, 2019-11-25:
I can confirm "pinch to zoom" is not working for me on Android 7, MediaTek device MT6735. The rest of the features works flawlessly, (not on 1.0.0-beta3.10 or 1.0.0-beta3.11)

android-dataticket, 2019-11-25:
> I can confirm "pinch to zoom" is not working for me on Android 7, MediaTek device MT6735. The rest of the features works flawlessly, (not on 1.0.0-beta3.10 or 1.0.0-beta3.11)
> 
> EDIT: Sorry I just saw it was noted that it'll make an appearance from beta4 on wards.

pretty sure they abandoned this project.. i had to add my own forked pinch to zoom over a year ago

Corical, 2019-11-26:
Would it be possible to push your implementation ?

android-dataticket, 2019-11-26:
> Would it be possible to push your implementation ?

pull request #72 

they never pushed it, but I've been using it for a couple years now at least

Corical, 2019-11-27:
Thanks, appreciated.