CameraKit/camerakit-android #451: Camera preview is very different than the native camera app

abhinav011085, 2018-11-13:
Native camera image
[image]
Camerakit's camera image
[image]

emersoncloud, 2018-11-13:
Thanks for creating this issue! Could you provide some more information to help us get to the bottom of this?

1. CameraKit Version (which v1.0.0 beta version are you using)
2. Android Device
3. Android Version
4. XML code where you create the CameraKitView 
5. And how you initialize the CameraKitView in Java

austinkettner, 2018-11-19:
Can you share your gradle implementation @abhinav011085 ? It still not clear which exact version of CameraKit you are using.

abhinav011085, 2018-11-20:
implementation group: 'com.camerakit', name: 'camerakit', version: '1.0.0-Snapshot', changing: true

emersoncloud, 2019-01-06:
Could be related to #470.

@abhinav011085 could you try on the latest release, `v1.0.0-beta3.10` and report back if the issue persists?

emersoncloud, 2019-01-29:
Any word on results from latest beta, `v1.0.0-beta3.11`? Closing this issue for now. Comment back if you're still encountering issues.

closed 2019-01-29 by emersoncloud