CameraKit/camerakit-android #452: RECORD_AUDIO premition on the premition list

ShalevCohen2112, 2018-11-13:
there is a way to remove the declaration of microphone premition from the application premition list?
 I am only using the picture function.

austinkettner, 2018-11-13:
We will make this use case is properly outlined is the new documentation. You can track the progress here: https://github.com/CameraKit/camerakit-android/issues/453

emersoncloud, 2018-11-14:
Closing in favor of #453.

closed 2018-11-14 by emersoncloud

emersoncloud, 2019-01-04:
Hi @ShalevCohen2112 as I mentioned in #453 we don't have a way to remove permissions from the permission list. CameraKit is designed to support camera, and video functionality and we need all 3 permissions to achieve that. 

But only the `AndroidManifest.xml` file will need to be adjusted if you want to fork and remove the permissions yourself!