CameraKit/camerakit-android #454: OnePlus 3 creates image with corruption on one edge

akiller, 2018-11-15:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
OnePlus 3 running OxygenOS 5.0.6 (Android 8.0.0).

CameraKit beta 3.1 and 3.9 both have this problem, I haven't tested all the releases in-between.

### Steps to Reproduce

Take a photo and save to storage. As far as I can tell other phones and the Android Emulator don't exhibit this issue.

I'm saving the image from camera.captureImage() directly to storage without any further processing.

### Expected Behavior

An image without the corruption.

### Actual Behavior

Photo taken in landscape. Notice the corruption on the right hand side of the photo:

[image]

Photo taken in portrait. Same corruption on the right hand side:

[image]

If I take the corrupted part, flip it, and move it over to the left, you can see it lines up with where it should be. I've left a white line to demonstrate the bit I've moved:

[image]

The same seems to be true for the second image,

### Reproducible Demo

I can make one if needed, but I'm just calling camera.captureImage() and saving the image directly so it will likely be the same in your own demo app.

Thanks

thubalek-inloop, 2018-11-26:
I can confirm this bug on many other devices too. I don't think it is device specific.

thubalek-inloop, 2018-11-27:
I also found root cause. The problem is caused by JpegKit.

There is following code in `com.camerakit.CameraPreview#capturePhoto`

```kotlin
val jpeg = Jpeg(it)
jpeg.rotate(captureOrientation)
```

`jpeg.rotate` busts the image. I'll also file bug in JpegKit project.

thubalek-inloop, 2018-11-27:
Created related ticked https://github.com/CameraKit/jpegkit-android/issues/4

akiller, 2018-11-27:
Ahh, good find. Hopefully if it occurs on multiple devices it'll be easier for someone with far more jpeg/NDK experience than myself to fix.

thubalek-inloop, 2018-11-27:
@akiller I'm currently digging into JpegKit and it seems to be bug either in JpegKit or directly in https://github.com/libjpeg-turbo/ (JpegKit is wrapper around jpeg-turbo library). I'll update this bug report with my findings.

snkashis, 2019-01-15:
@thubalek-inloop @akiller  yeah, this is definitely a multiple device problem. @austinkettner @emersoncloud is this serious issue on the docket for v1.0.0-beta3.11? I'm seeing it in beta3.9 and beta3.10.

PR #498 (2019-03-24): v1.0.0-beta3.12
Bug fixes and updates to CameraKit `v1.0.0-beta`

* Update JpegKit version. Closes #454