CameraKit/camerakit-android #455: Camera Preview streches on Rotation

usmanrana07, 2018-11-19:
Hi! I'm facing preview stretch issue. The issue produces like this: The screen is in Portrait mode , i lock the device to sleep mode, onStop method calls and i stop the camera in that. Then i rotate the device to landscape mode and unlock the device. in onResume method i start camera again. The preview is very stretched now.   Note: The screenOrientation of activity is set to Portrait.

emersoncloud, 2018-11-19:
Thanks for this issue! To help us better diagnose it, could you provide some more information?

1. CameraKit Version
2. Android Device
3. Android Version
4. XML and Java snippets where you initialize the CameraKitView

Thanks!

emersoncloud, 2018-11-19:
Closing this issue. I'll reopen when I hear back from you! =)

closed 2018-11-19 by emersoncloud

usmanrana07, 2018-11-19:
Using the latest CameraKit version. 
Device is Samsung S6 with version 7.0 (issue is on Nexus 5 and Note 8 as well and i believe is on all devices).

There is nothing fancy done at code part than simple commands to start or stop camera. The scenario to produce issue is described above.

HBiSoft, 2019-05-21:
Same issue on my Samsung J7 Pro running 8.0.1. It happens on the demo app as well.