CameraKit/camerakit-android #456: Don't works on Android Things

brunozaranza, 2018-11-21:
I tried do it works with different CK versions (0.12.0, 0.13.0, 1.0) and I have gotten different problems. In the latest version, CK is not able to connect to native camera of the Raspberry Pi 3. Someone have some tip how I might to do?

winterDroid, 2018-12-17:
Did you find a solution for it?

emersoncloud, 2018-12-17:
CameraKit has not been tested on the Raspberry Pi 3. I'm afraid theres not much we can do to help. Our development is focused on mainstream Android devices for the time being.

We hope to expand CameraKit to all connected devices sometime down the road. Closing this ticket for the time being, but hope to reopen and address later on. 

If you're able to find any solutions we'd love to take a look. It might help others running into similar problems on Raspberry Pis!

closed 2018-12-17 by emersoncloud