CameraKit/camerakit-android #458: NoClassDefFoundError with coroutines

jewom, 2018-11-24:
A lot of users have this error : 

**Caused by java.lang.NoClassDefFoundError kotlinx.coroutines.experimental.ThreadPoolDispatcherKt** 

`com.camerakit.CameraPreview.<init> (CameraPreview.kt:80)
com.camerakit.CameraKitView.obtainAttributes (CameraKitView.java:268)
com.camerakit.CameraKitView.<init> (CameraKitView.java:237)
java.lang.reflect.Constructor.constructNative (Constructor.java)
java.lang.reflect.Constructor.newInstance (Constructor.java:423)
android.view.LayoutInflater.createView (LayoutInflater.java:594)
android.view.LayoutInflater.createViewFromTag (LayoutInflater.java:696)
android.view.LayoutInflater.rInflate (LayoutInflater.java:755)
android.view.LayoutInflater.inflate (LayoutInflater.java:492)
android.view.LayoutInflater.inflate (LayoutInflater.java:397)
android.view.LayoutInflater.inflate (LayoutInflater.java:353)
android.support.v7.app.AppCompatDelegateImpl.setContentView (AppCompatDelegateImpl.java:469)
android.support.v7.app.AppCompatActivity.setContentView (AppCompatActivity.java:140)
com.belami.essential.activities.apps.camera.CameraActivity.onCreate (CameraActivity.kt:31)
android.app.Activity.performCreate (Activity.java:5264)
android.app.Instrumentation.callActivityOnCreate (Instrumentation.java:1088)
android.app.ActivityThread.performLaunchActivity (ActivityThread.java:2302)
android.app.ActivityThread.handleLaunchActivity (ActivityThread.java:2390)
android.app.ActivityThread.access$800 (ActivityThread.java:151)
android.app.ActivityThread$H.handleMessage (ActivityThread.java:1321)
android.os.Handler.dispatchMessage (Handler.java:110)
android.os.Looper.loop (Looper.java:193)
android.app.ActivityThread.main (ActivityThread.java:5292)
java.lang.reflect.Method.invokeNative (Method.java)
java.lang.reflect.Method.invoke (Method.java:515)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:824)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:640)
dalvik.system.NativeStart.main (NativeStart.java)`

[image]

[image]

My Gradle dependencies :
`dependencies {
    implementation fileTree(include: ['*.jar'], dir: 'libs')
    implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk7:$kotlin_version"
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
    implementation 'com.android.support.constraint:constraint-layout:1.1.3'
    implementation 'com.google.android.gms:play-services-auth:16.0.0'
    implementation 'com.android.support:appcompat-v7:28.0.0'

    implementation 'com.android.support:recyclerview-v7:28.0.0'

    implementation "org.jetbrains.anko:anko-common:0.10.4"

    implementation 'com.airbnb.android:lottie:2.5.1'
    implementation 'id.zelory:compressor:2.1.0'
    implementation('com.crashlytics.sdk.android:crashlytics:2.9.5@aar') { transitive = true }

    implementation 'com.google.firebase:firebase-core:16.0.3'
    implementation 'com.google.firebase:firebase-messaging:17.3.2'
    implementation 'com.google.android.gms:play-services-location:15.0.1'

    annotationProcessor "android.arch.persistence.room:compiler:1.1.1"
    kapt "android.arch.persistence.room:compiler:1.1.1"
    implementation "android.arch.persistence.room:runtime:1.1.1"

    implementation 'com.github.KwabenBerko:OpenWeatherMap-Android-Library:v1.1.1'

    implementation 'com.squareup.picasso:picasso:2.71828'
    implementation 'com.squareup.retrofit2:retrofit:2.4.0'
    implementation 'com.squareup.retrofit2:adapter-rxjava2:2.3.0'
    implementation 'com.squareup.retrofit2:converter-gson:2.4.0'
    implementation 'com.github.franmontiel:PersistentCookieJar:v1.0.1'
    implementation 'io.reactivex.rxjava2:rxandroid:2.1.0'

    implementation 'com.chibatching.kotpref:kotpref:2.5.0'
    implementation 'com.chibatching.kotpref:initializer:2.5.0'

    implementation 'de.hdodenhof:circleimageview:2.2.0'

    implementation 'com.camerakit:camerakit:1.0.0-beta3.9'
    implementation 'com.camerakit:jpegkit:0.1.0'

    implementation 'com.muddzdev:styleabletoast:2.1.2'
    implementation 'com.github.david-serrano:locationprovider:1.2'

    implementation 'com.android.support:multidex:1.0.3'
    implementation 'io.github.kobakei:ratethisapp:1.2.0'

    testImplementation 'junit:junit:4.12'
    androidTestImplementation 'com.android.support.test:runner:1.0.2'
    androidTestImplementation 'com.android.support.test.espresso:espresso-core:3.0.2'
}`

And this is my CameraActivity : 
`class CameraActivity : AppCompatActivity() {
    
    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_camera)

        btn_back.setOnClickListener { finish() }

        btn_switch.setOnClickListener {
            if (cameraView.facing == CameraKit.FACING_BACK)
                cameraView.facing = CameraKit.FACING_FRONT
            else
                cameraView.facing = CameraKit.FACING_BACK
        }

        btn_take_picture.setOnClickListener {

            cameraView.captureImage { _: CameraKitView, bytesImg: ByteArray ->

                    if (intent.getBooleanExtra(IS_ACTIVITY_WITH_RESULT, true)){
                        btn_send.visibility = View.VISIBLE
                        btn_save.visibility = View.GONE
                        btn_share.visibility = View.GONE
                    }
                    else{
                        btn_send.visibility = View.GONE
                        btn_share.visibility = View.VISIBLE
                    }

                    val img = BitmapFactory.decodeByteArray(bytesImg, 0, bytesImg.size)
                    preview.setImageBitmap(img)
                    previewVisibility()
                    btn_save.setOnClickListener {
                        ImagesTool.insertImage(contentResolver, img)
                        finish()
                    }
                    btn_cancel.setOnClickListener {
                        cameraVisibility()
                    }
                    btn_share.setOnClickListener {
                        val uri = MediaStore.Images.Media.insertImage(contentResolver, img, System.currentTimeMillis().toString(), "")
                        startActivity<ShareSelectContactActivity>(Intent.EXTRA_STREAM to uri)
                    }
                    btn_send.setOnClickListener {
                        val uri = MediaStore.Images.Media.insertImage(contentResolver, img, System.currentTimeMillis().toString(), "")
                        val i = Intent()
                        i.data = Uri.parse(uri)
                        i.addFlags(Intent.FLAG_GRANT_WRITE_URI_PERMISSION)
                        setResult(Activity.RESULT_OK, i)
                        finish()
                    }
            }
        }
    }

    override fun onStart() {
        super.onStart()
        cameraView.onStart()
    }

    override fun onResume() {
        super.onResume()
        cameraView.onResume()
    }

    override fun onPause() {
        cameraView.onPause()
        super.onPause()
    }

    override fun onStop() {
        cameraView.onStop()
        super.onStop()
    }

    private fun cameraVisibility(){
        preview.visibility = View.GONE
        toolbar.visibility = View.GONE
        cameraView.visibility = View.VISIBLE
        btn_take_picture.visibility = View.VISIBLE
    }

    private fun previewVisibility(){
        preview.visibility = View.VISIBLE
        toolbar.visibility = View.VISIBLE
        cameraView.visibility = View.GONE
        btn_take_picture.visibility = View.GONE
    }

}
`

emersoncloud, 2018-11-27:
@jewom Thanks for letting us know. I've updated the Kotlin and Kotlin coroutine versions in CameraKit v1.0.0-beta3.10. That should solve the issue you see here. Let me know otherwise. Thanks again!

austinkettner, 2018-12-22:
We will take a look. Though to note, `1.0.0-beta3.11-expensify` should not be used as it will not be sticking around. We will be issuing a proper `beta-3.11` shortly

emersoncloud, 2019-01-04:
@DummyCo @jewom With CameraKit **v1.0.0-beta3.10** you should use the following dependencies:
```
dependencies {
    implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
}
```

With CameraKit **v1.0.0-beta3.9**, use these dependencies instead: 
```
dependencies {
    implementation 'com.camerakit:camerakit:1.0.0-beta3.9'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.2.61'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:0.24.0'
}
```

If I was using beta3.10 and did not upgrade to the latest kotlin and coroutines I was able to reproduce the errors shown above. Those dependencies should solve your issues!

closed 2019-01-04 by emersoncloud

kinsleySafety360, 2019-04-20:
i have used both scenarios but i still get the error when i create a release apk version at times some of the version i had used i am failing to even create the release apk because of versions error ,  is anyone going through this and also has found a  solution to this  error , because i am now reconsidering my camera options .