CameraKit/camerakit-android #463: Proguard issues with AndroidX annotations

rohankandwal, 2018-12-03:
### Is this a bug report?

No

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

(Write your steps here:)

1. Add the library.
2. Add proguard rules in proguard file
3. Create a release build.

### Actual Behavior
Warning: com.camerakit.CameraKitView: can't find referenced class androidx.annotation.NonNull
Warning: com.camerakit.CameraKitView$Size: can't find referenced class androidx.annotation.NonNull
and so on..

I tried adding 
-dontwarn androidx.annotation.**
which allows app installation but the library crashes with following error :-
No pending exception expected: java.lang.NoSuchFieldError: no "J" field "nativeHandle" in class "Lcom/camerakit/preview/CameraSurfaceView;" or its superclasses

rohankandwal, 2018-12-03:
I resolved it while creating the issue but still posted the question since others might face same issue. Resolved it by replacing the proguard config by following codes :-

-dontwarn com.google.android.gms.**
-dontwarn androidx.annotation.**
-keep class com.camerakit.preview.* { *; }

The above code works instead of what mentioned in readme

closed 2018-12-03 by rohankandwal

emersoncloud, 2018-12-03:
@rohankandwal Can you comment with the dependencies section of your gradle file? I wasn't able to reproduce the issue with this configuration: 
```
implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
implementation 'io.alterac.blurkit:blurkit:1.0.0'
implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
implementation 'com.camerakit:jpegkit:0.1.0'
```

Just making sure I can reproduce your issue before changing our documentation. Thanks!