CameraKit/camerakit-android #467: captureImage callback not being called

UmarBhutta, 2018-12-09:
After updating my project with new version(1.0.0-beta3.10 and 3.9) of camerakit and migrating to kotlin 1.3.11 for and coroutines 1.0.0 rapid calling for the image capture callback is being lost. 
if you try to call captureImage method very quickly like clicking very fast. You will not get any callback for that click. Any suggestion?

qmetzler-luna, 2018-12-12:
It can happen even when not clicking fast, just calling it several time in a row will break the captureImage randomly.
By breaking I mean the callback is not called anymore and the camera also doesn't make the auto focus.
I tried calling
camera.onPause()
camera.onStop()
camera.onStart()
camera.onResume()
and camera.captureImage again but the callback is still not called.

1. Pixel 3
2. 9 API 28
3. 
```  
<com.camerakit.CameraKitView
            android:id="@+id/camera"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:fitsSystemWindows="true"
            android:keepScreenOn="true"
            app:camera_focus="continuous"
            app:camera_imageJpegQuality="100"
            app:camera_imageMegaPixels="2.0"
            app:camera_permissions="camera"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent" />
```
```
override fun onStart() {
    super.onStart()
    camera.onStart()
}
override fun onResume() {
    super.onResume()
    camera.onResume()
}
private fun setupCamera() {
    camera.errorListener = CameraKitView.ErrorListener { _, cameraException ->
        Log.e("Taking picture", "error")
    }
    button.setOnClickListener {
        Log.e("Taking picture", "trying to take a picture")
        camera.captureImage { _, image ->
            Log.e("Taking picture", "callback called")
            imageCaptured(image)
        }
    }
}
```

Thank you!

frogermcs, 2018-12-13:
Confirming - Pixel 3XL. Android 9. When I capture a couple times in a row, no callbacks are called from CameraKit. Not from captureImage, nor errorListener.

Interesting fact - when you stack "clicks", take app to background and bring it back, callback is delivered with the latest captured image.

Here are some random logs pointing on CamX:
```
2018-12-13 18:09:45.651 1164-1223/? E/libprocessgroup: Error encountered killing process cgroup uid 99085 pid 8676: No such file or directory
2018-12-13 18:09:46.599 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10157/pid_11303/memory.stat open failed: No such file or directory
2018-12-13 18:09:47.189 12521-12732/? E/Spotify: [Picasso-] Failed to loaded image:  -> content://com.spotify.music/image/
2018-12-13 18:09:47.582 12521-12521/? E/Spotify: [main] Failed to load image with uri: "".
    java.lang.IllegalArgumentException: Expected URL scheme 'http' or 'https' but no colon was found
        at aavo.a(SourceFile:1325)
        at aavn.f(SourceFile:916)
        at aawa.a(SourceFile:165)
        at gro.a(SourceFile:121)
        at yyk.a(SourceFile:44)
        at zdm.a(SourceFile:206)
        at zdm.run(SourceFile:159)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:458)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
        at zfd.run(SourceFile:411)
2018-12-13 18:09:47.617 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10027/pid_8982/memory.stat open failed: No such file or directory
2018-12-13 18:09:47.926 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10024/pid_11258/memory.stat open failed: No such file or directory
2018-12-13 18:09:53.156 12933-12933/? E/tunein.player: Not starting debugger since process cannot load the jdwp agent.
2018-12-13 18:09:53.418 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10073/pid_11869/memory.stat open failed: No such file or directory
2018-12-13 18:09:53.811 12933-12933/? E/tunein.player: Invalid ID 0x00000000.
2018-12-13 18:09:56.744 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10168/pid_11444/memory.stat open failed: No such file or directory
2018-12-13 18:10:09.084 799-2141/? E/libperfmgr: Failed to write to node: /sys/devices/system/cpu/cpu4/cpufreq/scaling_max_freq with value: 1996800, fd: 19
2018-12-13 18:10:09.117 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10179/pid_9136/memory.stat open failed: No such file or directory
2018-12-13 18:10:09.164 821-821/? E/lowmemorykiller: /dev/memcg/apps/uid_10031/pid_8068/memory.stat open failed: No such file or directory
2018-12-13 18:10:09.428 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:09.976 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:09.991 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:10.008 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:10.024 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:10.042 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:10.058 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:10.074 623-623/? E/libc: Access denied finding property "vendor.debug.egl.swapinterval"
2018-12-13 18:10:10.075 785-1282/? E/native: easel_manager_callback.cc:35 COMMAND_PROCESSOR_3PA on easel exited0
2018-12-13 18:10:10.131 785-1282/? E/libc: Access denied finding property "ro.camera.req.fmq.size"
2018-12-13 18:10:10.132 785-1282/? E/libc: Access denied finding property "ro.camera.res.fmq.size"
2018-12-13 18:10:10.160 785-1282/? E/CamX: [ERROR][HAL    ] camxchicontext.cpp:3826 CloneNodeProperties() No memory allocated for pNodeProperties
2018-12-13 18:10:10.160 785-1282/? E/CamX: [ERROR][HAL    ] camxchicontext.cpp:3826 CloneNodeProperties() No memory allocated for pNodeProperties
2018-12-13 18:10:10.161 785-1282/? E/GOOGLE_CHINODE_HDRPLUS: CHINODECAPSINFO is smaller than expected
2018-12-13 18:10:10.162 785-13462/? E/libc: Access denied finding property "persist.gcam.debug"
2018-12-13 18:10:10.162 785-13462/? E/libc: Access denied finding property "persist.gcam.save_text"
2018-12-13 18:10:10.162 785-13462/? E/libc: Access denied finding property "persist.gcam.save_metering"
2018-12-13 18:10:10.162 785-13462/? E/libc: Access denied finding property "persist.gcam.save_payload"
2018-12-13 18:10:10.162 785-13462/? E/CameraMetadataUtils: InitializeLensGcamGeometricCalibration: 335: distortion is NULL.
2018-12-13 18:10:10.163 785-1282/? E/CamX: [ERROR][HAL    ] camxchicontext.cpp:3826 CloneNodeProperties() No memory allocated for pNodeProperties
2018-12-13 18:10:10.163 785-1282/? E/CamX: [ERROR][HAL    ] camxchicontext.cpp:3826 CloneNodeProperties() No memory allocated for pNodeProperties
2018-12-13 18:10:10.189 785-1282/? E/CamX: [ERROR][HAL    ] camxmetadatapool.cpp:1600 SetMetadataByTag() Invalid slot; cannot set metadata tag 80080003
2018-12-13 18:10:10.189 785-1282/? E/CamX: [ERROR][HAL    ] camxmetadatapool.cpp:1763 PublishMetadataList() Invalid slot, cannot publish metadata list
[... cut ...]
2018-12-13 18:10:22.716 797-797/? E/NxpHal: write_unlocked failed - PN54X Maybe in Standby Mode - Retry
2018-12-13 18:10:22.737 2506-2506/? E/libnfc_nci: [ERROR:NativeNfcTag.cpp(801)] nativeNfcTag_doDisconnect: tag already deactivated
2018-12-13 18:10:22.769 13831-13831/? E/roid.apps.scon: Not starting debugger since process cannot load the jdwp agent.
2018-12-13 18:10:22.772 2565-2565/? E/LibSecureUISvc: svc_sock_send_message(suisvc): invalid remote socket suitch
2018-12-13 18:10:22.772 2565-2565/? E/LibSecureUISvc: svc_sock_send_message(suisvc): invalid remote socket suilst
```

frogermcs, 2018-12-13:
XML
```
<com.camerakit.CameraKitView
    android:id="@+id/cameraView"
    android:layout_width="match_parent"
    android:layout_height="wrap_content"
    android:layout_gravity="center|top"
    android:adjustViewBounds="true"
    app:camera_aspectRatio="1" />
```

Implementation:
```
public class MainActivity extends ClassifierActivity {

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        cameraView = findViewById(R.id.cameraView);
    }

    @Override
    protected void onStart() {
        super.onStart();
        cameraView.onStart();
    }

    @Override
    protected void onResume() {
        super.onResume();
        cameraView.onResume();
    }

    @Override
    protected void onPause() {
        cameraView.onPause();
        super.onPause();
    }

    @Override
    protected void onStop() {
        cameraView.onStop();
        super.onStop();
    }

    @Override
    public void onRequestPermissionsResult(int requestCode, String[] permissions, int[] grantResults) {
        super.onRequestPermissionsResult(requestCode, permissions, grantResults);
        cameraView.onRequestPermissionsResult(requestCode, permissions, grantResults);
    }

    public void onDetectClick(View v) {
        cameraView.captureImage((cameraKitView, picture) -> onNewImageByteArray(picture));
    }

    public void onNewImageByteArray(byte[] pictureBytesArray) {
        //...
    }
}
```

MaximeMaravalLunabee, 2018-12-20:
Same problem here. 
After import the library as module in my project, I finally avoid this issue by change 

```
@Synchronized
    override fun capturePhoto(callback: (jpeg: ByteArray) -> Unit) {
        this.photoCallback = callback

        if (cameraFacing == CameraFacing.BACK) {
            lockFocus()
        } else {
            captureStillPicture()
        }
    }
```

by 

```
@Synchronized
    override fun capturePhoto(callback: (jpeg: ByteArray) -> Unit) {
        this.photoCallback = callback

        if (cameraFacing == CameraFacing.BACK) {
            captureStillPicture()
        } else {
            captureStillPicture()
        }
    }
```

in `Camera2.kt`, but this is clearly not a good solution. 

Any news @emersoncloud  ? 
Thanks

UmarBhutta, 2018-12-20:
[image]
@emersoncloud it is happening on Samsung J7 pro.

DummyCo, 2018-12-22:
Same here, Pixel XL

huntj88, 2019-02-07:
I was able to fix this issue with this commit, based off the 1.0.0-beta3.11 branch
https://github.com/huntj88/camerakit-android/commit/837897cc0a623bfbf45046bc7999bd8b97f8bc41

huntj88, 2019-02-08:
> Same problem here.
> After import the library as module in my project, I finally avoid this issue by change
> 
> ```
> @Synchronized
>     override fun capturePhoto(callback: (jpeg: ByteArray) -> Unit) {
>         this.photoCallback = callback
> 
>         if (cameraFacing == CameraFacing.BACK) {
>             lockFocus()
>         } else {
>             captureStillPicture()
>         }
>     }
> ```
> 
> by
> 
> ```
> @Synchronized
>     override fun capturePhoto(callback: (jpeg: ByteArray) -> Unit) {
>         this.photoCallback = callback
> 
>         if (cameraFacing == CameraFacing.BACK) {
>             captureStillPicture()
>         } else {
>             captureStillPicture()
>         }
>     }
> ```
> 
> in `Camera2.kt`, but this is clearly not a good solution.
> @MaximeMaravalLunabee I think my fix also fixes what you were doing
> Thanks

huntj88, 2019-02-12:
last comment in my fork. The solution is to manually request the permissions yourself, rather than relying on the library to do it for you.

> So i played around with this a bit more, and I actually think the permission request code is busted. If i manually go grant the permission the library works without modification.
> 
> in the current code (as in not my fork), the permission request dialog is never presented, but it is in my fork.
> 
> Feel free to close this if you want

glenatron, 2019-02-21:
That CaptureStillImage from both the front and back camera fixes this problem for me also, but I assume there is a good reason it isn't supposed to work that way. Looks like when `lockFocus` is called it getting through to the `captureCallback.process` section going down the `STATE_WAITING_LOCK` path and in a situation where the afState is neither null, nor one of the focused_locked variants it hit the first `waitingFrames++` and then never seemed to re-enter the callback.

Interestingly when I closed my phone screen the callback immediately got triggered repeatedly, first resulting in ramping up the `waitingFrames` count until that triggered `captureStillPicture() and then continuing to be called afterwards, looping now endlessly through the `runPreCaptureSequence` cycle in a way I would expect to eat through battery.

gerdablum, 2019-02-27:
same issue on Nokia 7 plus (Android 9). With the code below the callback is triggered very randomly: 

(camerakit:1.0.0-beta3.11, jpegkit:0.1.0)

     override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_camera)
        camera_kit_view.focus = CameraKit.FOCUS_AUTO
        capture_fab.setOnClickListener {
            camera_kit_view.captureImage(object: CameraKitView.ImageCallback {
                override fun onImage(p0: CameraKitView?, byteImage: ByteArray?) {
                    Toast.makeText(this@CameraScanActivity,"You captured an image!", Toast.LENGTH_LONG).show()
                    scanQRCode(byteImage);
                }
            })
        }
    }

Rottinson, 2019-02-28:
Same issue on Xiaomi 8, Android 9. This problem occurs after a few normal calls. Once the callback is not called, it will never be called again unless you re-enter the app.
(camerakit:1.0.0-beta3.11, jpegkit:0.1.0)

XML:
```
<com.camerakit.CameraKitView
        android:id="@+id/main_camerakitview"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:adjustViewBounds="true"
        android:keepScreenOn="true"
        app:camera_facing="back"
        app:camera_flash="auto"
        app:camera_focus="continuous"
        app:camera_permissions="camera">
</com.camerakit.CameraKitView>
```
Java code:
```
@Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        cameraKitView = findViewById(R.id.main_camerakitview);
        takephotoButton = findViewById(R.id.main_takephotobutton);
        takephotoButton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                cameraKitView.captureImage(new CameraKitView.ImageCallback() {
                    @Override
                    public void onImage(CameraKitView cameraKitView, byte[] bytes) {
                        Bitmap capturedBitmap = BitmapFactory.decodeByteArray(bytes, 0, bytes.length);
                        capturedImageView.setImageBitmap(capturedBitmap);
                        String resultString = runModel(capturedBitmap);
                        resultTextView.setText(initialCapital(resultString));
                        progressBar.setVisibility(View.GONE);
                    }
                });
            }
        });
```

azwar, 2019-03-13:
Any update about this issue?

theFrantic, 2019-03-15:
Same problem here. Any update with this??

austinkettner, 2019-03-15:
We are not ignoring this, and plan to be back in touch by next week. This is one of our biggest bugs we plan to resolve.

theFrantic, 2019-03-15:
Very nice, thank you for the update and also for the great work, this library is great!

Tsabary, 2019-04-09:
Same here, capture Image is called very randomly OnePlus 5t

prikshit-hora, 2019-04-19:
Hi any updates on this issue

ggfan, 2019-04-19:
sorry to ask, is this one already fixed in 3.12-expensify?

alexcutovoi, 2019-04-23:
I have the same issue on HMD devices. Specifically HMD Global TA-1052.
Android 9.

fabio-delorenzo-wowza, 2019-05-17:
same issue

pecet66, 2019-05-23:
Samsung Galaxy S8, Android 9 - same issue.

BeaYeah, 2019-05-28:
> Same problem here.
> After import the library as module in my project, I finally avoid this issue by change
>
> ```
> @Synchronized
>     override fun capturePhoto(callback: (jpeg: ByteArray) -> Unit) [...]

¿How did you manage to import the library as a module?

jjames19951, 2019-06-21:
Same issue here, very frustrating and I can't seem to find a way around it

Twinsens, 2019-06-25:
Any updates for this issue?

roya90, 2019-06-26:
I have the same issue with Samung S7.

leoassuncao, 2019-06-26:
Same issue here with Samsung S10.
Any updates about it?

roya90, 2019-06-26:
I think it might be because of the type of threading for me. I called it within asynctask, I'm trying to use Handler instead.

roya90, 2019-06-27:
It works with Handler

marcosdecris, 2019-06-27:
@roya90 Can you show an exaple please?

roya90, 2019-06-27:
I changed back to using the regular camera, but I found this link very helpful. https://www.youtube.com/watch?v=adPLIAnx9og

alxgord, 2019-07-01:
Essential Phone PH-1
Android 10
same issue

jamolkhon, 2019-07-19:
Downgrading to beta3.10 worked for me.

jalacardio, 2019-07-20:
> Downgrading to beta3.10 worked for me.

Same, it works on my emulator now.

jtn-d, 2019-09-09:
Same here with Nexus 5 and beta3.11.

roya90, 2019-09-09:
@jtn-d go back to the original camera. Check out this link https://www.youtube.com/watch?v=adPLIAnx9og

EmilyChai, 2019-09-10:
Same here with Xiaomi 8, Android 9 and beta3.10
Is there any stable version, I just use the capture function.

azwar, 2019-09-10:
This is very long long issue and still not resolved. I do, have the same problem.

korbi-oranssi, 2019-09-14:
> We are not ignoring this, and plan to be back in touch by next week. This is one of our biggest bugs we plan to resolve. Thanks for the healthy discussion around this!

The same problem on Honor 20. We are waiting for the fix for 6 months already. Is that such a giant issue?

giacomodeglinnocenti, 2019-09-17:
Same issue here, it pretty a major issue.

Same effect with focus=continuous or focus=auto, no callback ever call to really take the photo as the library propose to.

Downgrading to 3.10 by:

   ```
 implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
```

denizsubasi, 2019-09-19:
> Same issue here, it pretty a major issue.
>
> Same effect with focus=continuous or focus=auto, no callback ever call to really take the photo as the library propose to.
>
> Downgrading to 3.10 by:
>
> ``` [...]

i have same issue.  -> Downgrading to 3.10

mocfaisal, 2019-09-30:
> Same issue here, it pretty a major issue.
>
> Same effect with focus=continuous or focus=auto, no callback ever call to really take the photo as the library propose to.
>
> Downgrading to 3.10 by:
>
> ``` [...]

Thanks

jomasim, 2019-10-16:
> Same issue here, it pretty a major issue.
>
> Same effect with focus=continuous or focus=auto, no callback ever call to really take the photo as the library propose to.
>
> Downgrading to 3.10 by:
>
> ``` [...]

Thanks, this also worked for me

leadersm, 2019-12-06:
when i downgrade to beta3.10, i got this error :

`W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
W/LegacyRequestMapper: Only received metering rectangles with weight 0.
    Only received metering rectangles with weight 0.
W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
    Only received metering rectangles with weight 0.
    Only received metering rectangles with weight 0.`

longld103, 2020-01-31:
I have fixed in my case when remove this
`@Override
    public void onRequestPermissionsResult(int requestCode, String[] permissions,
            int[] grantResults) {
        super.onRequestPermissionsResult(requestCode, permissions, grantResults);
        cameraKitView.onRequestPermissionsResult(requestCode, permissions, grantResults);
    }`

ElyDantas, 2020-03-08:
Couldn't take a picture using this library even with all answers above :[

XPHuster, 2020-05-06:
same issue

HMY314, 2020-05-29:
Nokia6 Android9  is same issue

luizfilho00, 2020-06-23:
Xiaomi Redmi Note 8, same issue

tuupainen, 2021-03-18:
According to my problem, there is multiple activities open and in background. LG Nexus's prints error; multiple camera clients active. It is difficult to really destruct activities from background, tried many methods. Hopefully .onClose() really frees all allocated resources. Some explicit destroy()-method would be nice.