CameraKit/camerakit-android #470: Front camera captures dark image with ckMethod=still

usmanrana07, 2019-01-01:
In Huawei Nova 3 device the front camera with ckMethod = still is capturing image as dark.

usmanrana07, 2019-01-03:
Device is 'Huawei Nova 3'   , captureMethod is 'still' , Android version is 7+ , CameraKit version is currently available at github.   

One more thing is that FLASH_ON doesn't work while capturing image with captureMethod 'still', but works for standard. For 'still' it flashes on screen tap but not while taking picture.

shadygoneinsane, 2019-01-08:
referred this https://camerakit.io/getstarted and am facing the same issue with front cam.. please fix this soon

emersoncloud, 2019-01-30:
@shadygoneinsane @usmanrana07 thanks for the feedback. I'm having trouble reproducing this on my own device. Just to make sure we're on the same page, does it happen for you when building and running the CameraKit demo application? You can find it inside the project living on the current master branch.

Thanks for the help!

usmanrana07, 2019-01-30:
@emersoncloud  it is very easily predicable atleast in the devices in which i've checked Samsung S6 (7.0) , Nova 3, Note5, Note8. Make sure you tried with with ckMethod=still .  And yes it is happening in CameraKit demo application as well i've verified and then posted the bug here.

prakaz, 2019-02-07:
> In Huawei Nova 3 device the front camera with ckMethod = still is capturing image as dark.

Which version of CameraKit are you using to be able to use ckMethod = still? is it 1.0.0-beta3.10 or 1.0.0-Snapshot?

usmanrana07, 2019-02-07:
I've  not added dependency but added it as a module and code is the most recent one that is available at github. Seems v0.13.5

nullptr2this, 2019-03-26:
I have been able to reproduce this issue consistently on a Pixel 3 while holding the phone in portrait perfectly perpendicular to the ground.  I can't speak to what the issue is, but the PreviewCallback implementation given to the setOneShotPreviewCallback call receives a completely nulled out byte[] for the frame every single time this occurs.  I'm wondering if this isn't a Camera 1 API bug.

I'm on a Pixel 3, Android 9, using CameraKit 0.13.5

[image]