CameraKit/camerakit-android #471: Convert Object to Video

nirpatel15, 2019-01-02:
I am trying to capture and save video using below method.

```
cameraView.captureVideo(new CameraKitView.VideoCallback() {
                                    @Override
                                    public void onVideo(CameraKitView cameraKitView, Object object) {
                                        videoCaptured(object);
                                    }
                                });
```
I get a java Object in parameter of **onVideo** . How can I convert it to a File or any other relevant entity..??

emersoncloud, 2019-01-02:
@nirpatel15 check out our documentation on capturing video in v0.13.2 [here](https://camerakit.io/docs?v=0.13.2#capturingvideo). 

You could also do something like the following to get the video file form the captureVideo function.
```
cameraView.captureVideo(new CameraKitEventCallback<CameraKitVideo>() {
    @Override
    public void callback(CameraKitVideo event) {
        event.getVideoFile();
    }
});
```

nirpatel15, 2019-01-03:
I am using version beta 3.10 and I am unable to find such event in this version. Do I need to downgrade..??

emersoncloud, 2019-01-03:
Yes you would need to downgrade if you want to use video in your application. Video is not yet supported in the latest v1.0.0 beta versions but will be in the near future. We don't have a solid time estimate but we are working on it.

In the mean time you can use CameraKit v0.13.2,  the last version with video support.

closed 2019-01-03 by emersoncloud

HBiSoft, 2019-05-21:
Unfortunately v0.13.2 has this bug #384 (explained in detail at the bottom of the page).

HBiSoft, 2019-11-03:
> how even to record a video in 0.13.2, I'm getting video as object in the callback, how to use this object

Have a look at [this](https://github.com/natario1/CameraView) library instead.