CameraKit/camerakit-android #477: Crash in Api 16  lib 1.0.0-beta3.10

rodolfoizidoro, 2019-01-10:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Emulator Google Api 16

Project min api 16
Target sdk api 27
Compile api 27

### Steps to Reproduce

Use library 1.0.0-beta3.10

### Expected Behavior

Open Camera

### Actual Behavior

`01-10 12:10:02.312 4509-4509/ E/AndroidRuntime: FATAL EXCEPTION: main
    java.lang.RuntimeException: Unable to start activity ComponentInfo{.MaskCameraActivity}: android.view.InflateException: Binary XML file line #11: Error inflating class com.camerakit.CameraKitView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2059)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2084)
        at android.app.ActivityThread.access$600(ActivityThread.java:130)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1195)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:137)
        at android.app.ActivityThread.main(ActivityThread.java:4745)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:511)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
        at dalvik.system.NativeStart.main(Native Method)
     Caused by: android.view.InflateException: Binary XML file line #11: Error inflating class com.camerakit.CameraKitView
        at android.view.LayoutInflater.createView(LayoutInflater.java:613)
        at uk.co.chrisjenx.calligraphy.CalligraphyLayoutInflater.createCustomViewInternal(CalligraphyLayoutInflater.java:211)
        at uk.co.chrisjenx.calligraphy.CalligraphyLayoutInflater.access$000(CalligraphyLayoutInflater.java:20)
        at uk.co.chrisjenx.calligraphy.CalligraphyLayoutInflater$PrivateWrapperFactory2.onCreateView(CalligraphyLayoutInflater.java:302)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:680)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:746)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:489)
        at uk.co.chrisjenx.calligraphy.CalligraphyLayoutInflater.inflate(CalligraphyLayoutInflater.java:60)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:396)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:352)
        at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287)
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
        at MaskCameraActivity.onCreate(MaskCameraActivity.kt:35)
        at android.app.Activity.performCreate(Activity.java:5008)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1079)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2023)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2084) 
        at android.app.ActivityThread.access$600(ActivityThread.java:130) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1195) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:137) 
        at android.app.ActivityThread.main(ActivityThread.java:4745) 
        at java.lang.reflect.Method.invokeNative(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:511) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553) 
        at dalvik.system.NativeStart.main(Native Method) 
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.constructNative(Native Method)
[... 78 lines cut ...]
        at android.view.LayoutInflater.inflate(LayoutInflater.java:396) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:352) 
        at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287) 
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139) 
        at MaskCameraActivity.onCreate(MaskCameraActivity.kt:35) 
        at android.app.Activity.performCreate(Activity.java:5008) 
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1079) 
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2023) 
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2084) 
        at android.app.ActivityThread.access$600(ActivityThread.java:130) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1195) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:137) 
        at android.app.ActivityThread.main(ActivityThread.java:4745) 
        at java.lang.reflect.Method.invokeNative(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:511) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553) 
        at dalvik.system.NativeStart.main(Native Method) `

[image]

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

closed 2019-01-23 by austinkettner

emersoncloud, 2019-01-23:
Hey @rodolfoizidoro thanks for letting us know about this issue! We've added the fixes to both v1.0.0-beta3.11 as well as 0.13.5. We've made your changes ourselves to prevent the need for a Contributor License Agreement. We truly appreciate your contributions to CameraKit, without your PR we wouldn't have made these changes. 

Go ahead and give either of the versions a try! Thanks again.

patrickjason91, 2019-03-12:
I am using the `1.0.0-beta3.10` version of CameraKit lib. I used a Sony Xperia M (Android API 16, 4.1.x) and I experienced the same issue.

```
<com.camerakit.CameraKitView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="wrap_content"
            android:adjustViewBounds="true"
            app:camera_facing="back"/>
```

What worked for me was to do the same fix commented by @4brunu from #256
I added the following lines to the Activity where I am using the `CameraKitView`
```
static {
        System.loadLibrary("camerakit-core");
    }
```

PR #487 (2019-01-23): v0.13.5
## CameraKit v0.13.5
This patch fixes JPEG transformer loading issue when capturing an image on API level 17 and below.

Closes #478 
Closes #477