CameraKit/camerakit-android #480: How i can use version v0.13.4?

mnayef95, 2019-01-11:
How i can use version v0.13.4?
i tried 

`implementation 'com.camerakit:camerakit:0.13.4'`

but it's doesn't work

emersoncloud, 2019-01-11:
To use v0.13.4, the gradle line is: 

```implementation 'com.wonderkiln:camerakit:0.13.4'```

The `v0.13.X` line is under `wonderkiln` and the new `v1.0.0-betaX` releases are all under `camerakit`. Sorry for the confusion!

closed 2019-01-11 by emersoncloud

KryptKode, 2019-11-06:
Can this be included in the documentation? It could be confusing.

snaqviAndroidApp, 2020-01-15:
Hi 

I am trying to use video recording api available (if any), however getting the following situation: 

I am getting the following when trying in both of these: 

1. implementation 'com.camerakit:camerakit:0.13.4'  (error: not able to resolve... )
2. implementation 'com.camerakit:camerakit:1.0.0-0.13.4' (same error)

more in the docs (https://camerakit.io/docs), no matter what version I choose: the docs are shown for: 1.0.0-beta3.11

Please if it could be helped

regards

rahadurr, 2020-03-31:
> To use v0.13.4, the gradle line is:
> 
> `implementation 'com.wonderkiln:camerakit:0.13.4'`
> 
> The `v0.13.X` line is under `wonderkiln` and the new `v1.0.0-betaX` releases are all under `camerakit`. Sorry for the confusion!

Thanks It's work for me..

PetroHrynyuk, 2023-10-03:
Failed to resolve: com.wonderkiln:camerakit:0.13.4