CameraKit/camerakit-android #485: CameraKit take a lot of time to enter in cameraView.captureImage

jewom, 2019-01-21:
Hi,

I am using com.camerakit:camerakit:1.0.0-beta3.10 and when I take a picture with a lot of light, it take a lot of time to enter in the captureImage listener.
When I put my finger front of the camera it's working instant!

Why ? Wan I fix the delay ?

fabrizziocht, 2019-01-21:
Same error using a Samsung galaxy S6 with android 6.0.1 API 23. I tested with API 19, 21, 22 and 27, and works well, except this version.

jewom, 2019-01-22:
So it depend of the API of the device ?

Hyperbolas, 2019-02-15:
Sam problem in ASUS Zenfone4, which is API 26.

ControlAdad, 2019-02-16:
Same problem on Samsung tablet, API 27 (8.1.0)

WajidAli007, 2019-12-09:
Samsung Tab, when there is light, it will either a lot of time.
If its taking time, put a finger on the camera and immediately remove it (removing is not necessary) the callback will be called. I don't know why is it so......