CameraKit/camerakit-android #486: Huawei mate9pro opens the photo page and spends the screen.

bluesky0506, 2019-01-22:
### Is this a bug report?

华为mate9pro打开拍照页面，花屏。无法使用。必显，谢谢！
Huawei mate9pro opens the photo page and spends the screen. Not available. Must show, thank you!
I am sorry, my english lever is poor. add issue picture, think you !
链接: https://pan.baidu.com/s/1yX2Ga3JYPQpUXWBrHgv0Xw 提取码: aq3y

I used version is newest:
implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
implementation 'com.camerakit:jpegkit:0.1.0'
implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

(Write your answer here.)

### Environment

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

AnyLifeZLB, 2019-01-23:
I have the same problem

[image]

you can upload the picture by just Ctrl+c --> Ctrl+v

bluesky0506, 2019-01-24:
thank you, let me try.
[image]

bluesky0506, 2019-01-24:
This issure on front camera

emersoncloud, 2019-01-27:
Duplicate of #454. We'll post updates there as they come. Closing this issue. Thanks!

closed 2019-01-27 by emersoncloud