CameraKit/camerakit-android #492: Caused by: android.view.InflateException: Binary XML file line #8: Error inflating class com.wonderkiln.camerakit.CameraView

jcsofts, 2019-01-26:
implementation 'com.wonderkiln:camerakit:0.13.4'

2019-01-26 19:54:15.087 24153-24153/com.yuserapp E/unknown:ReactNative: Exception in native call
    android.view.InflateException: Binary XML file line #8: Binary XML file line #8: Error inflating class com.wonderkiln.camerakit.CameraView
    Caused by: android.view.InflateException: Binary XML file line #8: Error inflating class com.wonderkiln.camerakit.CameraView
    Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:334)
        at android.view.LayoutInflater.createView(LayoutInflater.java:658)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:801)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:741)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:874)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
        at com.yuserapp.camera.ReactCameraView.init(ReactCameraView.java:107)
        at com.yuserapp.camera.ReactCameraView.<init>(ReactCameraView.java:91)
        at com.yuserapp.camera.CameraViewManager.createViewInstance(CameraViewManager.java:76)
        at com.yuserapp.camera.CameraViewManager.createViewInstance(CameraViewManager.java:46)
        at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:42)
        at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:223)
        at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:152)
        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:1012)
        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:983)
        at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
        at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:134)
        at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:105)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:979)
        at android.view.Choreographer.doCallbacks(Choreographer.java:790)
        at android.view.Choreographer.doFrame(Choreographer.java:718)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:967)
        at android.os.Handler.handleCallback(Handler.java:809)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:166)
        at android.app.ActivityThread.main(ActivityThread.java:7383)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:469)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:963)
     Caused by: java.lang.NoSuchFieldError: No static field surface_view_container of type I in class Lcom/wonderkiln/camerakit/core/R$id; or its superclasses (declaration of 'com.wonderkiln.camerakit.core.R$id' appears in /data/app/com.yuserapp-DwfHZN_IcCyMb5VCHbHXSw==/base.apk!classes3.dex)
        at com.wonderkiln.camerakit.SurfaceViewPreview.<init>(SurfaceViewPreview.java:27)
        at com.wonderkiln.camerakit.CameraView.<init>(CameraView.java:132)
        at com.wonderkiln.camerakit.CameraView.<init>(CameraView.java:99)

I use react native, the 0.13.2 is works fine for me, but get this issue when replace to 0.13.4

jcsofts, 2019-01-27:
0.13.5 have the same issue

emersoncloud, 2019-01-27:
@jcsofts that's great news! Thanks for bringing this issue to our attention.

closed 2019-01-27 by emersoncloud

yashashree-web, 2020-05-18:
my app is crashing as soon as it is launched due to inflating pronlem of wonderkiln camerakit how am i supposed to correct this error

2020-05-18 12:20:07.885 11795-11795/com.google.android.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.google.android.cameraview.demo, PID: 11795
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.google.android.cameraview.demo/com.google.android.cameraview.demo.objetCameraActivity}: android.view.InflateException: Binary XML file line #10: Binary XML file line #10: Error inflating class com.wonderkiln.camerakit.CameraView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2988)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3066)
        at android.app.ActivityThread.-wrap11(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1764)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:202)
        at android.app.ActivityThread.main(ActivityThread.java:6962)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:528)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:888)
     Caused by: android.view.InflateException: Binary XML file line #10: Binary XML file line #10: Error inflating class com.wonderkiln.camerakit.CameraView
     Caused by: android.view.InflateException: Binary XML file line #10: Error inflating class com.wonderkiln.camerakit.CameraView
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:334)
        at android.view.LayoutInflater.createView(LayoutInflater.java:650)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:797)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:737)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:870)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:831)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:555)
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:161)
        at com.google.android.cameraview.demo.objetCameraActivity.onCreate(objetCameraActivity.java:22)
        at android.app.Activity.performCreate(Activity.java:7227)
        at android.app.Activity.performCreate(Activity.java:7218)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1214)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2941)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3066)
        at android.app.ActivityThread.-wrap11(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1764)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:202)
        at android.app.ActivityThread.main(ActivityThread.java:6962)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:528)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:888)
     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.wonderkiln.camerakit.SurfaceViewContainer.addOnLayoutChangeListener(android.view.View$OnLayoutChangeListener)' on a null object reference
        at com.wonderkiln.camerakit.SurfaceViewPreview.<init>(SurfaceViewPreview.java:28)
        at com.wonderkiln.camerakit.CameraView.<init>(CameraView.java:132)
        at com.wonderkiln.camerakit.CameraView.<init>(CameraView.java:99)
        at java.lang.reflect.Constructor.newInstance0(Native Method) 
        at java.lang.reflect.Constructor.newInstance(Constructor.java:334) 
        at android.view.LayoutInflater.createView(LayoutInflater.java:650) 
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:797) 
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:737) 
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:870) 
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:831) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:515) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:423) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:374) 
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:555) 
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:161) 
        at com.google.android.cameraview.demo.objetCameraActivity.onCreate(objetCameraActivity.java:22) 
        at android.app.Activity.performCreate(Activity.java:7227) 
        at android.app.Activity.performCreate(Activity.java:7218) 
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1214) 
[... cut ...]
        at android.app.ActivityThread.-wrap11(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1764) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:202) 
        at android.app.ActivityThread.main(ActivityThread.java:6962) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:528) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:888)