CameraKit/camerakit-android #494: The preview view show incorrect ratio after stop video in landscape orientation

jcsofts, 2019-01-27:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

(Write your steps here:)
I test the demo of camerakit. 

1. first hold down the the video button in landscape orientation
2. release the video button to stop the video
3. the preview will be scale distortion

### Expected Behavior

preview will be scale distortion

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

hksfho, 2019-02-24:
same as you

tyoc213, 2019-06-06:
same on some phones

ErAmrit, 2019-07-30:
Yes, I am also experiencing the same issue, there is any clue to fix it.

FuherProgramer, 2019-10-03:
me too ! but why no body answer us and fix this big issue