CameraKit/camerakit-android #503: Is there a way to prevent screen orientation and use it in a specific mode(i.e. Portrait mode) for capturing image and videos.

shahzebakansari, 2019-02-05:


emersoncloud, 2019-02-06:
Hey @shahzebakansari thanks for the question. There is not a method for locking the CameraView apart from the general application locking done through the [Android Manifest file](https://stackoverflow.com/questions/4675750/lock-screen-orientation-android). 

What's your use case for locking the Camera without locking the application?

shahzebakansari, 2019-02-08:
The use case for locking the camera is for capturing image or video in a specific dimension ratio like 4:3 (Portrait mode).
And yes by locking the activity orientation via Manifest file does not solve the problem as the internal functioning of camera i.e screen orientation still works and not overriden.

pribeh, 2019-02-24:
Looking for a solution to this as well and can't see anything in the documentation in regards to it.

manujjagga421, 2019-03-13:
@shahzebakansari  @pribeh Any update on this? I also wish to achieve same( click images/videos in only portrait).