CameraKit/camerakit-android #506: Camera block after captureImage

mattianavacchia, 2019-02-12:
Hi, after taking one shot with captureImage method, I can't free the camera and the system is stucked. How can i do?

emersoncloud, 2019-02-14:
Hey @mattianavacchia thanks for the issue. Can you please give us some more detail about your CameraKit implementation? Specifically:

1. Android Version
2. CameraKit Version
3. It would be great to see a quick video demo of the issue if you are able.

Thanks!

slobglob, 2019-02-18:
Same here with your latest v1.0.0-beta
Running on Android 7, Hwuei p10.
No video yet, but it's how you imagine it to be, capturing an image and the screen freezes.

Fazalcs13, 2019-02-27:
> Hi, after taking one shot with captureImage method, I can't free the camera and the system is stucked. How can i do?

I am also facing the same issue.

Brand:  Sony
Model: Xperia XZ Premium Dual G8142
Also known as Sony Xperia XZ Premium Dual-Sim, Sony Maple DS