CameraKit/camerakit-android #507: Distorted screen on the CameraView - Version 0.13.5

roip890, 2019-02-19:
Hi,

I am using in version 0.13.5.
I have a distortion screen on the CameraView when I try to take a photo from the front camera.
Is there a chance you fixed that problem on 1.0.0-beta version?
And if so, can you try to help to fix that problem on version 0.13.5?
Video camera is essential for me, and I know that it isn't supported on version 1.0.0-beta.

Thanks,
Roi.

nullptr2this, 2019-03-23:
I have a fix for this.  PR coming shortly.

nullptr2this, 2019-03-25:
@roip890 I have a pull request up to fix this issue.  Camera preview sizing options are not always set because of an orientation conversion issue.  Hopefully this is approved/accepted, but if not you can check out the code here: https://github.com/CameraKit/camerakit-android/pull/521

commit 2624ab90 (2019-03-25): Merge pull request #521 from nullptr2this/preview-distortion-fix

Fix preview rendering distortion issues in some scenarios

- Close #507
closed 2019-03-25 by austinkettner