CameraKit/camerakit-android #510: Working perfectly in Android 4, but not work in Android 8. Same code.

SlonGL, 2019-02-24:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Two phones.
API 17 (4.2.2) - everything working perfectly 
API 27 (8.1.0) - In onCreate - togleFacing() working, but in fun called with button onClick - no. CaptureImage - same.

### Steps to Reproduce

start app, call activity with camera. in 4.2 - ok, in 8.1 - no. 
Permissions
Manifest.permission.READ_EXTERNAL_STORAGE,
Manifest.permission.WRITE_EXTERNAL_STORAGE,
Manifest.permission.CAMERA
asked in manifest for 4.2 and asked in app for 8.1

Same issue for emulator API 28

### Expected Behaviour

Photo onClick photo button, and toggle facing onCllck toggle button.

### Actual Behaviour

 public fun rotateCameraButtonPressed(view : View) {
        cameraKitView?.toggleFacing() 
        // has no effect, but works ok when called from onCreate.
        // Is called when press button, I checked it with breakpoint in debug
 }

Hasiy, 2019-03-15:
我也有相同的问题但将代码改成 如下似乎没问题了
on xml
`   <com.camerakit.CameraKitView
                    android:id="@+id/camera_kit"
                    android:layout_width="match_parent"
                    android:layout_height="wrap_content"
                    android:adjustViewBounds="true"
                    android:keepScreenOn="true"
            />`

on kotlin
`  private fun setCameraKit() {
        cameraKitView = findViewById(R.id.camera_kit)
        cameraKitView.facing = FACING_BACK
        cameraKitView.flash = FLASH_OFF
        cameraKitView.focus = FOCUS_AUTO
    }`

KaelMa, 2019-05-13:
> ### Is this a bug report?
> Yes
>
> ### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
> Yes
>
> ### Environment
> Two phones.
> API 17 (4.2.2) - everything working perfectly
> API 27 (8.1.0) [...]

same as you , on Android 8.0 not work

KaelMa, 2019-05-13:
same as @SlonGL .
@Hasiy 's solution not work.

KaelMa, 2019-05-13:
update: 
Pixel 1 (Android 9) works perfectly
Mi 6 (Android 8) not work