CameraKit/camerakit-android #512: The Camera2 implementation sometimes tries to use the CameraDevice after it is closed.

glenatron, 2019-03-01:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Sony Xperia G8441, but judging by the source code it should happen across the board.

### Steps to Reproduce

In the Camera2 implementation it is possible to call `captureStillPicture` in a situation where the `cameraDevice` is closed, I have not been able to find a consistent reproduction for this, but it crashes my app from time to time.

### Expected Behavior

If the cameraDevice is closed, it should drop the capture request.

### Actual Behavior

The following exception is thrown, and we crash out:

> java.lang.IllegalStateException: CameraDevice was already closed
>         at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2445)
>         at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureRequest(CameraDeviceImpl.java:773)
>         at com.camerakit.api.camera2.Camera2.captureStillPicture(Camera2.kt:199)
>         at com.camerakit.api.camera2.Camera2.capturePhoto(Camera2.kt:155)

### Reproducible Demo

The behaviour is inconsistent- in my case it appears to be caused by a race condition, but there is nothing in the code to mitigate it, setting a `cameraConnected` property from the `CameraDevice.StateCallback()` events is the standard approach to resolving this problem.

glenatron, 2019-03-19:
Having worked with this a lot more, on my app the problem traditionally shows when I have taken a picture ( which takes the user to a different activity ) then switched back to the activity containing the CameraPreview. That seems to cause the image to freeze almost immediately very consistently. The most common error message that shows up in my logs around this happening is: `E/libc: Access denied finding property "vendor.camera.aux.packagelist"` - this doesn't seem to get flagged up anywhere I can access it as a consumer of the API but once that has shown up in the logs the camera is disconnected and the API does not seem  able to recover it automatically.

vininandu, 2019-05-31:
Same issue I am also facing, I am using RTSP library for streaming and I am able to stream video in lower versions and in latest 9PKQ1.180904.001 version and child version it is not working and throwing following error
Access denied finding property "vendor.camera.aux.packagelist"
Access denied finding property "vendor.camera.hal1.packagelist"