CameraKit/camerakit-android #513: CameraKitView.FrameCallback is never getting called

darwinfrancis, 2019-03-04:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

(Write your steps here:)

1. add    the below dependencies
    implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.11'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
2. prepare CameraKitView in Activity
3. implement CameraKitView.FrameCallback 
4. call  cameraKitView.captureFrame(frameCallback); in button click

### Expected Behavior

When I click on the button the camera kit api should provide each frames as byte[] in below callback
```
@Override
        public void onFrame(CameraKitView cameraKitView, byte[] bytes) {
            showLog("on frame");
        }
```

### Actual Behavior

The CameraKitView.FrameCallback is never getting called.

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

GaidamakUA, 2019-03-26:
That's a big issue. I need it to scan QRCode or BarCode, but I can't. Because API isn't implemented.

bpappin, 2019-03-26:
It would be very handy to be able to plug in APIs like the Google vision API, or zxing.