CameraKit/camerakit-android #514: Picture captured from front cam is showing a thin layer of back cam view at right corner

usmanrana07, 2019-03-06:
In Note 5 when i capture picture from front cam then it shows a thin layer of back camera view on right corner of image.

Unfreeze-kk, 2019-03-20:
Same issue in HUAWEI P20.

glenatron, 2019-03-28:
Same on a Sony Xperia, I thought it was just a pale line, but no it is clearly a little bit of an image. I make it 8 pixels wide on a 1080x1920 image.

helang1991, 2019-08-26:
Same issue in HUAWEI HONOR

chiel99, 2019-08-29:
Hi all,
In my case, I realized this issue caused by turbo-jpeg.
If the image's width or height is not evenly divisible by the MCU block size (which is 16 here), the cropping result will create odd-looking strips on the right or bottom edge of the image.

I fix it by adding `TJXOPT_TRIM` option to discard the partial MCU blocks.
Hope this helps.

```
diff --git a/camerakit/src/main/cpp/JniJpegTransformer.cpp b/camerakit/src/main/cpp/JniJpegTransformer.cpp
index 9da50aa..1c73283 100644
--- a/camerakit/src/main/cpp/JniJpegTransformer.cpp
+++ b/camerakit/src/main/cpp/JniJpegTransformer.cpp
@@ -148,7 +148,7 @@ Java_com_wonderkiln_camerakit_JpegTransformer_jniCommit
         cropRegion.h = jpegTransformer->cropHeight;

         transform->r = cropRegion;
-        transform->options = TJXOPT_CROP;
+        transform->options = TJXOPT_CROP | TJXOPT_TRIM;
     }
```

usmanrana07, 2019-09-19:
@chiel99  this solution didn't work for me

TrentPierce, 2019-09-30:
Also having this same issue on Samsung S5, S7, S9+. Basically anything ive tested on. How can we crop this?