CameraKit/camerakit-android #525: java.lang.RuntimeException at com.camerakit.api.camera2.a.a(Camera2.kt:48)

Hasiy, 2019-03-29:
出现问题的设备：GI-I9500_TMMARS    OS：5.1   还有 OPPO A3  VIVO X20  MI2S
`java.lang.RuntimeException
    at com.camerakit.api.camera2.a.a(Camera2.kt:48)
    at com.camerakit.api.f.a(ManagedCameraApi.kt:12)
    at com.camerakit.CameraPreview.a(CameraPreview.kt:234)
    at com.camerakit.CameraPreview$start$1$1.a(CameraPreview.kt:118)
    at kotlin.coroutines.jvm.internal.BaseContinuationImpl.b(ContinuationImpl.kt:32)
    at kotlinx.coroutines.bc$a.b(Dispatched.kt:235)
    at kotlinx.coroutines.az.run(Dispatched.kt:81)
    at kotlinx.coroutines.bl.e(EventLoop.kt:123)
    at kotlinx.coroutines.g.j(Builders.kt:69)
    at kotlinx.coroutines.j.a(Builders.kt:45)
    at kotlinx.coroutines.i.a(Unknown Source)
    at kotlinx.coroutines.j.a(Builders.kt:35)
    at kotlinx.coroutines.i.a(Unknown Source)
    at com.camerakit.CameraPreview$start$1.a(CameraPreview.kt:115)
    at kotlin.coroutines.jvm.internal.BaseContinuationImpl.b(ContinuationImpl.kt:32)
    at kotlinx.coroutines.bc$a.b(Dispatched.kt:235)
    at kotlinx.coroutines.az.run(Dispatched.kt:81)
    at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:422)
    at java.util.concurrent.FutureTask.run(FutureTask.java:237)
    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:152)
    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:265)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
    at java.lang.Thread.run(Thread.java:818)`

Hasiy, 2019-03-29:
CameraKit v1.0.0-beta3.10

rjsuzuki, 2020-06-01:
This issue is still occurring in v1.0.0-beta3.11

rajiraji123, 2021-06-24:
> 出现问题的设备：GI-I9500_TMMARS OS：5.1 还有 OPPO A3 VIVO X20 MI2S
> `java.lang.RuntimeException at com.camerakit.api.camera2.a.a(Camera2.kt:48) at com.camerakit.api.f.a(ManagedCameraApi.kt:12) at [...]

I am getting the same issue in OPPO A52.Any solution for this??