CameraKit/camerakit-android #532: Captured image resolution is different from preview resolution

Divay-Parashar, 2019-04-06:
### Is this a bug report?

(Yes)

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

(Yes)

### Environment
Android 
Motorola one power

### Steps to Reproduce

(Write your steps here:)

1. Add CameraKit library in your project and  complete basic setup of Library with your Activity.
2. capture image with function "cameraKitView.captureImage(new  CameraKitView.ImageCallback() " and save image somewhere in the device storrage.
3.Then compare Camera preview resolution and saver

### Expected Behavior

  Camera Preview Size  (width=1920, height=1080)
  Camera Image Captured Size(width=1920, height=1080)

### Actual Behavior

Camera Preview Size(width=1920, height=1080)
Camera Image Captured Size(width=2560, height=800)

ScreenShots  of Error.
camera preview when application open in device .

[image]

This is how image looks when i save this image in device storage ..

[image]

### Reproducible Demo

This is my  zipped project. 
[CameraTest.zip](https://github.com/CameraKit/camerakit-android/files/3050448/CameraTest.zip)
Plese give me solution as soon as possible my project is stuck just because of this issue.

Divay-Parashar, 2019-04-15:
Did you find any solution?

nullptr2this, 2019-04-18:
What version of the library are you using?

Divay-Parashar, 2019-04-19:
I'm using  'com.camerakit:camerakit:1.0.0-beta3.10' version but this issue is still in latest version '3.11'
and there are lots of bugs in the latest version that's why I'm using this version.

tinkooladik, 2019-04-22:
Hi there! Unfortunately I've got the same issue on some devices.
On screenshots I'm putting an ImageView with result picture above the Preview.

I also attached device info, but the problem accrued not only on this device.
I use 1.0.0-beta3.10 also tried 3.11, the same result

http://prntscr.com/nf82ny
http://prntscr.com/nf82xt
http://prntscr.com/nf833f

Divay-Parashar, 2019-04-22:
> Hi there! Unfortunately I've got the same issue on some devices.
> On screenshots I'm putting an ImageView with result picture above the Preview.
>
> I also attached device info, but the problem accrued [...]

Hi tinkooladik could you please suggest me another better camera library.

dendyaziz, 2019-06-25:
I found this problem happened on > 2000px res.
Just change output resolution to 1.9 or less.
    cameraKitView.setImageMegaPixels(1.9f);

tinkooladik, 2019-07-05:
@dendyaziz cool, it helped, thank you!

wangwei66666, 2021-02-18:
I encountered the same problem, the phone model is Huawei Honor 30, how to solve the problem？

DzulhafizAR, 2021-03-15:
Any way to resolve this other than using the imageMegaPixels? I found using that, the result are not the same for all devices.

fangor16, 2021-03-15:
Any way to resolve this other than using the imageMegaPixels? I found using that, the result are not the same for all devices.