CameraKit/camerakit-android #533: CaptureImage doesn't work most of the time on android

Tsabary, 2019-04-09:
### Is this a bug report?
(write your answer here)
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
Yes
(Write your answer here.)

### Environment

Android Studio emulator / OnePlus 5t

### Steps to Reproduce

(Write your steps here:)

1. Implement captureImage
2. Try to take a photo
3.

### Expected Behavior

(Write what you thought would happen.)

A photo would be taken

### Actual Behavior

(Write what happened. Add screenshots!)

The action doesn't kick in at all. I've entered a few Logs, the first being immedietly after calling to capture image, but whenever that issue exists (which is almost always for me both at the emulator and OnePlus device, that Log never prints.

### Reproducible Demo

https://stackoverflow.com/questions/55526362/capture-image-action-only-works-sometimes-without-clear-reason-why

You could see here the relevant part of the code. The log that should print "image captured" almost never prints (and a photo isn't taken). When a photo is taken, the log of course prints, but there is no reasoning behind when it would or wouldn't work. It seems that it is more likely to work on the emulator.

drogeek, 2019-04-17:
I've the same problem and have noticed it would take a picture if I obstruct the lens with for instance my finger.

Tsabary, 2019-04-18:
It wouldn't be a good solution, but it doens't work for me anyway.
I've been pushing this fix to the end because I really want to avoid changing to Fotoapparat (moved it from there to CameraKit before), but looks like I'll have to. Really hoping for an update soon that'll fix it.

Akshshr, 2019-04-18:
Yes, noticing the same...CaptureImage doesnt do anything at least on 
`1.0.0-beta3.11'`

Tsabary, 2019-04-18:
is there a previous version in which you know it did work? I don't mind downgrading for now, my camera request is super simple

arsenii-repository, 2019-04-18:
Experiencing same problem. Anyone have forced CameraKit work properly?

@Tsabary I also moved from Fotoapparat to CameraKit in hope it will work. (Fotoapparat seems doesn't work on API 17)

austinkettner, 2019-04-18:
Hey guys, can you be more specific about the versions of CK you are using? This isn't a known widespread issue but obviously is affecting you. We will look into it once we know more!

arsenii-repository, 2019-04-18:
build.gradle
```  implementation 'com.camerakit:camerakit:1.0.0-beta3.11'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
```

Also no errors/info in logcat. No error in CameraKitView.ErrorListener.

Akshshr, 2019-04-18:
Here are my details
Gradle
```
 implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
    implementation 'com.camerakit:jpegkit:0.1.0'
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
```

I think the root of a lot of these issues are that the sample app has very deprecated code. Would be nice to have that on V1

Tsabary, 2019-04-18:
I am using 3.11 now but the problem still existed when i was on 3.10.

I am running my app on the emulator and on a OnePlus 5t.

This is how my code is structured:

 `captureButton.setOnClickListener {

            Log.d("photoActivity", "button clicked")

            cameraKitView.captureImage(){ _, p1 ->
                Log.d("photoActivity", "image captured")
}
}`

I've also tried instead of `captureImage()` to have it `captureImage` or `captureFrame`. The current setp seems to be the most consistent and I was just able to make it work with my OnePlus as well (literally for the first time as I was writing this) but it is still not consistent.

button clicked is always printed, where image captured almost never does (it rarely happens and inly on the emulator).

Tsabary, 2019-04-18:
Update: as per Murphy's law, things might actually be working for me now. It seems to be more consistent after I've added the empty brackets. I am still running tests but it is deffinitely working much better than it did before.

Tsabary, 2019-04-26:
Update: it stops working again. I can't get it to be stable.
Before calling it an issue, I think it would be best if you can provide with the proper way to capture an image using Kotlin, because currently I can't tell if it is nor consistent because of the library or because I am doing something wrong (though I have tried quit a few different options)

Uplink03, 2019-04-27:
I was trying to debug this, but Android Studio claimed that the bytecode didn't match the source code for CameraKitView, so I've cloned branch _v1.0.0-beta4_, added the _camerakit_ directory to my project and, with some effort, made it build as a subproject just like the demo app does in this repository instead of using the pre-build release.

Good news: with the integrated project this works. So my guess is that the published build is out of date.

I'm not sure I'd be able to tell you how I did the subproject integration, as it was quite a bit of trial and error (I'm new to Android development), using the gradle files in this repository as reference each time I had an error.

leokvw, 2019-05-06:
Same problem here in my Google Pixel 1. I have implemented the captureImage function as instructed and Logcat didn't print anything strange...and that's strange hhhh.

gabrielbatta, 2019-05-06:
Same problem

tycallen, 2019-05-07:
same problem

wintermoon9, 2019-05-07:
Any insights? Dealing with this too...

fireb86, 2019-05-08:
same problem, only with 3.11

chintan-mishra, 2019-05-11:
Here is what I have found while testing on some devices:
Moto G4+:
- No image can be captured
- `toggleCamera()` fails

Xiaomi Mi A10: 
- TODO: I will update this within 24 hours.

OnePlus 6:
- Low light results in no image capture and `toggleCamera()` method doesn't work
- Proper lighting leads to one successful image capture however subsequent image callbacks are never triggered. The same happens with `toggleCamera()` method.
- When the device is placed on a flat surface then any number of photos can be captured. `toggleCamera()` works each time.

PORTB, 2019-05-12:
same problem, SM-G935FD

FelipeBerrios, 2019-05-15:
same problem here on Samsung Device ( Tab Active 2)

DarkMinstrel, 2019-06-07:
Surprisingly enough, adding an errorListener to the CameraKitView has fixed the problem.

chintan-mishra, 2019-06-10:
@DarkMinstrel it doesn't seem to work for some devices.

Intonjokel, 2019-06-13:
I have the same Problem. Sometimes it work. But most of the time it doesn't work.

Jaime97, 2019-06-27:
Same problem here

toeii, 2019-08-23:
Same problem, from MI 6(Android 8.0)

toeii, 2019-08-23:
> Same problem, from MI 6(Android 8.0)

This problem can be avoided by lowering the version to v1.0.0-beta3.10

simplepeng, 2019-10-09:
same problem from oneplus3A and oneplus6

nxnyk, 2019-10-19:
the same problem

javdev, 2019-11-24:
same for me. beta3.10
Works sometimes, doesn't work sometimes. Built in a retry with stopping starting the camera again. Works sometimes, doesn't work sometimes. Is this project discontinued. The last change seemed to be happening months ago...

javdev, 2019-11-24:
and 3.10 same problem. I even cleaned the project, rebuilt it, invalidated the caches and restarted.

sjd753, 2019-12-23:
Same problem with implementation 'com.camerakit:camerakit:1.0.0-beta3.11' and kotlin on android 10 it doesn't capture any image. No errors are logged or thrown in listener. Are they working on it or is it discontinued???

passatgt, 2019-12-23:
Downgrading to beta 3.10 solved the issue for me

javdev, 2019-12-23:
Hi, I think it is discontinued, because Google Jetpack has CameraX integrated. So it wouldn't make sense to try to compete with it. CameraX has image analysis on top.

cleancontrol, 2020-01-22:
Downgrading to beta 3.10 solved the issue for me

matteomedioli, 2020-02-12:
Downgrading to beta 3.10 also works for me on RedmiNote7!
```
implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
implementation 'com.camerakit:jpegkit:0.1.0'
implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.0'
implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'

tian102, 2020-10-14:
> > Same problem, from MI 6(Android 8.0)
> 
> This problem can be avoided by lowering the version to v1.0.0-beta3.10

Worked for me! Oh, happy days!

103945089, 2021-01-04:
> Downgrading to beta 3.10 solved the issue for me

worked for me too, Thanks

ArslanKhan99, 2021-01-13:
Thanks 

> 
> 
> > Downgrading to beta 3.10 solved the issue for me
> 
> worked for me too, Thanks

Thanks Work for me

ruben93, 2021-08-14:
In the new version you should activate microphone permissions to take a photo 🤷🏻‍♂️