CameraKit/camerakit-android #536: CameraKit latest gradle version's startVideo() captureVideo() implemention is empty

ourdex86, 2019-04-15:
### Is this a bug report?

(Yes)

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

(Yes)

### Environment
Android Studio : 3.3.2
Gradle Version : 4.10.1
Android Plugin Version : 3.3.2
camerakit version : 1.0.0-beta3.11

### Steps to Reproduce

(Write your steps here:)

`1.   <com.camerakit.CameraKitView
            android:id="@+id/cameraKitView"
            android:layout_width="0dp" android:layout_height="0dp"
            app:layout_constraintTop_toTopOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintBottom_toBottomOf="parent"
            android:adjustViewBounds="true"
            android:keepScreenOn="true"
            app:camera_flash="auto"
            app:camera_facing="back"
            app:camera_focus="continuous"
            app:camera_permissions="camera"
    >
`
2.From MainActivity.kt, cameraKitView.startVideo() or cameraKitView.captureVideo{ cameraKitView, any ->  }
3. Go inside the implementation of this method. Its empty

`   public void startVideo() {
    }

    public void stopVideo() {
    }

    public void captureVideo(CameraKitView.VideoCallback callback) {
    }

    public void captureFrame(CameraKitView.FrameCallback callback) {
    }

    public void setFrameCallback(CameraKitView.FrameCallback callback) {
    }`

### Expected Behavior

(It should capture the video.)

### Actual Behavior

(Nothing is happening.)

Muhaiminur, 2019-04-16:
Same case happening for me :(

austinkettner, 2019-04-18:
Hey guys, v1.x does not currently contain any video support. It is coming soon though!

closed 2019-04-18 by austinkettner

bstillitano, 2019-05-14:
@austinkettner do you have an ETA on video support? Would love to continue using your library but if it's not going to be out for some time will need to move on. Understand that you're doing all you can.

HBiSoft, 2019-05-21:
@austinkettner Can you please provide any timeline on when you plan on releasing it?