CameraKit/camerakit-android #537: v1.0.0-beta3.11- the front camera

leandro1995, 2019-04-20:
I've been testing camerakit in version 3.11 and I realize that in the version of Android 6 and 5 the front camera fails to select it with a button sometimes it turns all black or does not want to change the camera and that I used the CameraKit.FACING_FRONT with the setFacing and the result is the same I had to use version 3.10 that does not have that problem

KaelMa, 2019-05-16:
I found the problem is caused by this commit. After I revert it , it works.
[image]
[image]

closed 2023-05-10 by leandro1995

weeklynote, 2023-05-10:
已收到，我会尽快回复你！
                     ——刘劲字