CameraKit/camerakit-android #540: Stretched preview for front camera Samsung S10

dummy-amit, 2019-05-07:
### Is this a bug report?

No, I am facing issue with Samsung S10 device front camera

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

yes

### Environment

CameraKit Version: 0.13.4

Android Device: Samsung Galaxy s10

Android Version: 9

### Steps to Reproduce

1. Just start camera preview for front camera

### Expected Behavior

centre cropped preview

### Actual Behavior

Stretched preview.

[image]

xml attributes set

 <com.wonderkiln.camerakit.CameraView
        android:id="@+id/camera_view"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"
        android:adjustViewBounds="true"
        app:ckCropOutput="false"
        android:soundEffectsEnabled="false"
        app:ckDoubleTapToToggleFacing="false"
        app:ckFacing="back"
        app:ckFlash="off"
        app:ckFocus="continuous"
        app:ckJpegQuality="100"
        app:ckMethod="standard"
        app:ckPinchToZoom="true"
        app:ckVideoQuality="max720p"
        android:layout_width="match_parent"
        android:layout_height="match_parent"/>

olmur, 2019-06-11:
the same on s9

olmur, 2019-06-13:
Downgrade to 0.13.0, there is no such bug

AhmedZenhom, 2019-06-15:
> Downgrade to 0.13.0, there is no such bug

You're right there is no such bug for photo capturing in 0.13.0 ,  but it still exist in video capturing :/