CameraKit/camerakit-android #542: Camera preview black after initial permissions grant

barees63, 2019-05-09:
After granting permission the (back) camera preview remains black, if I toggle between front & back camera it starts working.

Once the initial permissions have been granted the preview works correctly every time.

I have CameraKitView hosted in a fragment which is in a ViewPager.  All lifecycle
events are being forwarded to camera-kit in the fragment.

This is reproducible in both simulator and physical devices, so far only tested on Android 9.

Using beta3.10

barees63, 2019-05-09:
OK, this issue for me was that the permissions result was being sent to the activity and I had to manually forward it to the fragment hosting the camera.

closed 2019-05-09 by barees63

abhilasha3297, 2019-09-23:
@barees63 how did you go about it?