CameraKit/camerakit-android #546: Android resource linking failed

linuxhsj, 2019-05-17:
`Android resource linking failed
Output:  E:\gp\CameraKit\app\src\main\res\layout\activity_main.xml:29: error: attribute camera_facing (aka com.camerakit.app:camera_facing) not found.
E:\gp\CameraKit\app\src\main\res\layout\activity_main.xml:29: error: attribute camera_focus (aka com.camerakit.app:camera_focus) not found.
E:\gp\CameraKit\app\src\main\res\layout\activity_main.xml:29: error: attribute camera_imageJpegQuality (aka com.camerakit.app:camera_imageJpegQuality) not found.
E:\gp\CameraKit\app\src\main\res\layout\activity_main.xml:29: error: attribute camera_imageMegaPixels (aka com.camerakit.app:camera_imageMegaPixels) not found.
E:\gp\CameraKit\app\src\main\res\layout\activity_main.xml:29: error: attribute camera_permissions (aka com.camerakit.app:camera_permissions) not found.` @eomine @eomine @muratgu

huasheng28, 2019-05-23:
I ran  the master branch code with the same error. and didnt find solution yet

ibrarHussain87, 2019-06-03:
i am also getting same error developer please guide us what is the main issue i applied many solution but did not work need help buddy

anthonynguyen194, 2019-06-08:
Same as well, would really like to test some modifications to this lib

Shiguangqiang, 2019-06-21:
The same question, is there a solution?

peterchaula, 2019-06-23:
@linuxhsj, @arsonistAnt and @Shiguangqiang, clone and check out the 1.0.0-beta branch. The problem was fixed that branch.

keofrol, 2019-08-06:
no, not fixed

lucky-you, 2019-10-21:
How to solve this problem？

Afra55, 2019-10-29:
Add a parameter declaration to attr.xml based on CameraKitView.java:
```
<declare-styleable name="CameraKitView">
        <attr name="android_adjustViewBounds" format="boolean" />
        <attr name="camera_aspectRatio" format="float" />
        <attr name="camera_facing" format="integer" />
        <attr name="camera_flash" format="integer" />
        <attr name="camera_focus" format="integer" />
        <attr name="camera_zoomFactor" format="float" />
        <attr name="camera_permissions" format="integer" />
        <attr name="camera_imageMegaPixels" format="float" />
        <attr name="camera_imageJpegQuality" format="integer" />
    </declare-styleable>
```

then change activity_main.xml:
```
 <com.camerakit.CameraKitView
                android:id="@+id/camera"
                android:layout_width="match_parent"
                android:layout_height="wrap_content"
                android:adjustViewBounds="true"
                app:camera_facing="0"
                app:camera_focus="1"
                app:camera_imageJpegQuality="100"
                app:camera_imageMegaPixels="2.0"
                app:camera_permissions="1" />
```
Delete classes that are not used and reported an error: CameraView.java.

manishakamble, 2020-03-20:
@Afra55  thank you so much its work for me

ganeshInnoctive, 2020-03-24:
@manishakamble Were you able to solve the problem using @Afra55's solution. I am unable to solve it at my end. Can you please help?

manishakamble, 2020-04-21:
@ganeshInnoctive . can I know the exceptions or errors you got?