CameraKit/camerakit-android #547: android.view.InflateException-crashing my app with this {Error Inflating class com.camerakit.cameraview }

Sangeeth7, 2019-05-19:
### Is this a bug report?

yes

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

No

### Environment

(Please include the following information along with any other relevant environment details.)

CameraKit Version: com.camerakit:camerakit:1.0.0-beta3.11

Android Device: mi note5 pro

Android Version: android 8.1 oreo

### Steps to Reproduce

(Write your steps here:)

1.I'm following camerakit documentation and i'm trying to run in my mobile
2.its crashing as soon as app is alunched
3.
repositories {
    maven { url 'https://dl.bintray.com/camerakit/camerakit-android'}
}
dependencies {
    implementation 'com.camerakit:camerakit:1.0.0-beta3.11'
}

#XML implementation

<?xml version="1.0" encoding="utf-8"?>
<FrameLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    xmlns:com.wonderkiln="http://schemas.android.com/apk/res-auto"
    android:id="@+id/activity_main"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:paddingBottom="16dp"
    android:paddingLeft="16dp"
    android:paddingRight="16dp"
    android:paddingTop="16dp"
    tools:context=".MainActivity">

    <com.camerakit.CameraKitView
    xmlns:app="http://schemas.android.com/apk/res-auto"
    android:id="@+id/cameraView"
    android:layout_width="300dp"
    android:layout_height="300dp"
    android:layout_gravity="center|top" />

</FrameLayout>

#MainActivity implementation
as in camerakit documentation

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

(Include your CameraKit setup and usage.)

Rishab2019, 2019-06-07:
Hello,

I was having the same issue.After changing the implementation to beta1, issue got resolved.

But the photo is not storing in the Internal storage.Any help?

rjsuzuki, 2020-01-23:
Try creating a ViewStub and inflate the CameraKitView object programmatically in the lifecycle onViewCreated+. Worked for me.