CameraKit/camerakit-android #549: CameraX support

brianHaoxiang, 2019-05-24:
Hi friends,

Google has released the CameraX and it seems it provides better and consistent APIs. https://developer.android.com/jetpack/androidx/releases/camerax

Is there any plan to leverage the CameraX?

crearo, 2019-06-12:
From what I see, there's nothing CameraX provides that CameraKit doesn't. Think of it like a parallel CameraKit developed by the folks at Google. 
The only thing I don't see in CameraKit is obtaining image buffers in CPU memory for every frame, which can be done using both the camera 1 and camera 2 apis easily.

Mostafayehya, 2019-07-21:
@crearo 
Acutally you can have access to every frame by using the Analysis use case, check this out : 
https://developer.android.com/training/camerax/analyze

crearo, 2019-07-22:
Nice! Yeah CameraX provides it, but I don't think CameraKit (this repo) does.

Mostafayehya, 2019-07-22:
I tried to, but couldn't find anything related to accessing the frames, ( maybe it exists ),but well hidden,Anyway accessing frames is easier with Camera API and CameraX .