CameraKit/camerakit-android #553: Surface had no valid native Surface on Nexus 7

KaviIDPal, 2019-06-18:
### Is this a bug report?
Yes
Getting crash randomly when opening CameraView.

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

I get a 406 when trying to read them

### Environment

CameraKit Version: 0.11.2

Android Device: Nexus 7

Android Version: 6.0

### Steps to Reproduce

1. While opening camera
2. App get crashed

### Expected Behavior

### Actual Behavior
App should not crash.

### Reproducible Demo
Crash Log:

Fatal Exception: java.lang.IllegalArgumentException: Surface had no valid
native Surface.
       at
android.hardware.camera2.legacy.LegacyCameraDevice.nativeGetSurfaceId(LegacyCameraDevice.java)
       at
android.hardware.camera2.legacy.LegacyCameraDevice.getSurfaceId(LegacyCameraDevice.java:658)
       at
android.hardware.camera2.legacy.LegacyCameraDevice.containsSurfaceId(LegacyCameraDevice.java:678)
       at
android.hardware.camera2.legacy.RequestThreadManager$2.onPictureTaken(RequestThreadManager.java:220)
       at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1092)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:148)
       at
android.hardware.camera2.legacy.CameraDeviceUserShim$CameraLooper.run(CameraDeviceUserShim.java:136)
       at java.lang.Thread.run(Thread.java:818)

KaviIDPal, 2019-06-18:
Can you please update the release date of v 1.0.0 ?

GowriIDPal, 2019-06-18:
For me also getting the same error