CameraKit/camerakit-android #557: Fatal Exception: java.lang.RuntimeException: Error during updateTexImage (see logcat for details)

Greeeeyyyss, 2019-07-16:
### Crash Log

`Fatal Exception: java.lang.RuntimeException: Error during updateTexImage (see logcat for details)
       at android.graphics.SurfaceTexture.nativeUpdateTexImage(SurfaceTexture.java)
       at android.graphics.SurfaceTexture.updateTexImage + 243(SurfaceTexture.java:243)
       at com.camerakit.preview.CameraSurfaceTexture.updateTexImage + 31(CameraSurfaceTexture.java:31)
       at com.camerakit.preview.CameraSurfaceView.onDrawFrame + 60(CameraSurfaceView.java:60)
       at android.opengl.GLSurfaceView$GLThread.guardedRun + 1571(GLSurfaceView.java:1571)
       at android.opengl.GLSurfaceView$GLThread.run + 1270(GLSurfaceView.java:1270)`

### Environment

CameraKit Version: 1.0.0-beta3.0

Android Device: Samsung S10+

Android Version: Android 9.0

I am using Kotlin and AndroidX.

Skyler-export, 2025-02-02:
Take a look at this https://lnkr.fun/cdd7c25 reply

weeklynote, 2025-02-02:
已收到，我会尽快回复你！
                     ——刘劲字