CameraKit/camerakit-android #559: java.lang.ClassNotFoundException: Didn't find class "jpegkit.JpegImageView" on path: DexPathList

ChrisB85, 2019-08-01:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

Yes

### Environment

Windows 10, Android Studio 3.4.2

CameraKit Version: 1.0.0-beta3.11

Android Device: Pocophone F1

Android Version: 9

### Steps to Reproduce

I have created a sample (empty project), so to reproduce it you need to:
1. Clone repo: https://github.com/ChrisB85/CameraTest.git
2. Compile, install & run :)

### Expected Behavior

It should not crash ;)

(Write what you thought would happen.)

### Actual Behavior

It crashes :(

`2019-08-01 20:33:36.739 23263-23263/test.test.camera E/AndroidRuntime: FATAL EXCEPTION: main
    Process: test.test.camera, PID: 23263
    java.lang.RuntimeException: Unable to start activity ComponentInfo{test.test.camera/test.test.camera.MainActivity}: android.view.InflateException: Binary XML file line #48: Binary XML file line #48: Error inflating class jpegkit.JpegImageView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2946)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3081)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1831)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:201)
        at android.app.ActivityThread.main(ActivityThread.java:6806)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:873)
     Caused by: android.view.InflateException: Binary XML file line #48: Binary XML file line #48: Error inflating class jpegkit.JpegImageView
     Caused by: android.view.InflateException: Binary XML file line #48: Error inflating class jpegkit.JpegImageView
     Caused by: java.lang.ClassNotFoundException: Didn't find class "jpegkit.JpegImageView" on path: DexPathList[[zip file "/data/app/test.test.camera-bZFe2cH_jPvn3EBIrMVAnw==/base.apk"],nativeLibraryDirectories=[/data/app/test.test.camera-bZFe2cH_jPvn3EBIrMVAnw==/lib/arm64, /data/app/test.test.camera-bZFe2cH_jPvn3EBIrMVAnw==/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:171)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at android.view.LayoutInflater.createView(LayoutInflater.java:610)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:794)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:734)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:867)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:870)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:870)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:519)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:427)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:469)
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:140)
        at test.test.camera.MainActivity.onCreate(MainActivity.java:52)
        at android.app.Activity.performCreate(Activity.java:7224)
        at android.app.Activity.performCreate(Activity.java:7213)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1272)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2926)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3081)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1831)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:201)
        at android.app.ActivityThread.main(ActivityThread.java:6806)
        at java.lang.reflect.Method.invoke(Native Method)
2019-08-01 20:33:36.739 23263-23263/test.test.camera E/AndroidRuntime:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:873)`

(Write what happened. Add screenshots!)

### Reproducible Demo

https://github.com/ChrisB85/CameraTest.git

(Paste the link to an example project and exact instructions to reproduce the issue.)

(Include your CameraKit setup and usage.)

luga97, 2019-08-15:
Hi, I had the same issue a while ago, you need to change the layout view from `jpegkit.JpegImageView` to `com.jpegkit.JpegImageView` and it's all :)

ChrisB85, 2019-08-18:
@luga97 Thank you!