CameraKit/camerakit-android #575: Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest android.hardware.camera2.impl.CameraDeviceImpl$CaptureCallbackHolder.getRequest(int)' on a null object reference

Greeeeyyyss, 2019-10-01:
Experiencing a crash on few users with devices below upon capturing an image. This issue is not reproducible on other devices like Samsung and Pixel. If it helps see logs:

`Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest android.hardware.camera2.impl.CameraDeviceImpl$CaptureCallbackHolder.getRequest(int)' on a null object reference
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onCaptureErrorLocked + 2137(CameraDeviceImpl.java:2137)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onDeviceError + 1827(CameraDeviceImpl.java:1827)
       at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage + 306(CameraDeviceUserShim.java:306)
       at android.os.Handler.dispatchMessage + 106(Handler.java:106)
       at android.os.Looper.loop + 164(Looper.java:164)
       at android.os.HandlerThread.run + 65(HandlerThread.java:65)`

**CameraKit Version:** camerakit:1.0.0-beta3.10

**Android Device:** LG Tribute Empire, LG Stylo 4, MIRO (Foxx)

**Android Version:** 8.1.0

williamycyh, 2020-01-10:
I also got the crash on same device (oppo and vivo)

nikomp, 2023-02-02:
I also got the crash on Xiaomi Redmi 5 Android 8.1.0