CameraKit/camerakit-android #577: Image not captured on beta 11 with a Samsung Galaxy A3

akiller, 2019-10-09:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

Yes

### Environment

CameraKit Version: v1.0.0beta3.11 and v1.0.0beta3.10

Android Device: Samsung Galaxy A3 (2017) - SM-A320SL

Android Version: 8.0.0

### Steps to Reproduce

1. Build/run CameraKit demo app
2. Click take photo

### Expected Behavior

Photo to be captured and the preview image displayed on the UI.

### Actual Behavior

The camera preview is still visible but the photo is never captured. On beta 10 the photo is eventually captured if I use my hand to cover up the camera so it goes black after clicking the capture button - it's very strange!

Here's a quick video demonstrating what I mean:
https://photos.app.goo.gl/Q4ms9u9sXUvVkwzc9

Stepping through the code in a debugger what I believe might be the root cause is that when CameraPreview.capturePhoto() is called it runs GlobalScope.launch with a callback. If I put a breakpoint on the callback it never gets hit, therefore, none of the capture logic gets ran.

As another test I put a breakpoint in the callback for it inside pause() and navigated to another app and that didn't get hit either.

Perhaps I'm missing something as Kotlin and coroutines are a bit of a mystery to me.

akiller, 2019-10-10:
I've been trying to get to the bottom of this all day, still no luck. However, if I set FORCE_DEPRECATED_API to true so it uses Camera 1 it seems to work fine, therefore it looks like it could be a Camera2 issue?

akiller, 2019-10-11:
Sorry for the comment spam but I've half got around this by introducing the ability for the application to manually override which camera API gets used, either by setting the attribute app:camera_apiVersion or using cameraView.setCameraApiVersion(). Setting these to auto defaults to your current behaviour for figuring out which to use, or you can use "camera1" or "camera2" to force whichever one you like.

I can't create a pull request because the beta11 branch appears to be deleted, and the develop branch appears to be broke and an older version of the library . So here is both a link to my repo with the changes on and a patch :

https://github.com/akiller/camerakit-android/tree/switch-camera-api

[switch-camera-api.diff](https://github.com/CameraKit/camerakit-android/files/3715096/switch-camera-api.txt)

I think allowing the app to switch the camera API out when something goes wrong is a good idea, but hopefully we can figure out why camera2 is broke in this instance too.

luga97, 2019-10-28:
I have same issue on Xiaomi Redmi 6A and Samsung Galaxy S7, i solved this problem using v1.0.0beta3.10

akiller, 2019-10-28:
I think we upgraded from 3.10 too which triggered it. We've also found this breaks on a Motorola 6.

A colleague of mine thinks he might have fixed the problem in 3.11. There were two issues he found within coroutines queues and the use of global scope which broke it.

We'll submit a pull request/patch if our testing proves this works, I expect it to take a week or so.

kiler222, 2019-12-22:
I used your patch and it works. Thanks a lot:)

I confirm that when I force set 'camera1' to be used then capturing images works fine. Once I switch to camera2 then it breaks.