CameraKit/camerakit-android #588: Sample Project Availble on this Project gradle build failed.

nawinkhatiwada, 2019-11-12:
Got this error when building the project after cloning.

Android Studio version: `3.5`

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugResources'.
> Android resource linking failed
  Output:  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: error: attribute camera_facing (aka com.camerakit.app:camera_facing) not found.
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: error: attribute camera_focus (aka com.camerakit.app:camera_focus) not found.
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: error: attribute camera_imageJpegQuality (aka com.camerakit.app:camera_imageJpegQuality) not found.
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: error: attribute camera_imageMegaPixels (aka com.camerakit.app:camera_imageMegaPixels) not found.
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: error: attribute camera_permissions (aka com.camerakit.app:camera_permissions) not found.
  error: failed linking file resources.
  
  Command: /Users/nabin/.gradle/caches/transforms-1/files-1.1/aapt2-3.2.1-4818971-osx.jar/0956a280f3a16fb98d7379b9aee64c63/aapt2-3.2.1-4818971-osx/aapt2 link -I\
          /Users/nabin/Library/Android/sdk/platforms/android-28/android.jar\
          --manifest\
          /Users/nabin/Desktop/camerakit-android/app/build/intermediates/merged_manifests/debug/processDebugManifest/merged/AndroidManifest.xml\
          -o\
          /Users/nabin/Desktop/camerakit-android/app/build/intermediates/processed_res/debug/processDebugResources/out/resources-debug.ap_\
          -R\
          @/Users/nabin/Desktop/camerakit-android/app/build/intermediates/incremental/processDebugResources/resources-list-for-resources-debug.ap_.txt\
          --auto-add-overlay\
          --java\
          /Users/nabin/Desktop/camerakit-android/app/build/generated/not_namespaced_r_class_sources/debug/processDebugResources/r\
          --custom-package\
          com.camerakit.app\
          -0\
          apk\
          --output-text-symbols\
          /Users/nabin/Desktop/camerakit-android/app/build/intermediates/symbols/debug/R.txt\
          --no-version-vectors
  Daemon:  AAPT2 aapt2-3.2.1-4818971-osx Daemon #0
  Output:  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: AAPT: error: attribute camera_facing (aka com.camerakit.app:camera_facing) not found.
      
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: AAPT: error: attribute camera_focus (aka com.camerakit.app:camera_focus) not found.
      
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: AAPT: error: attribute camera_imageJpegQuality (aka com.camerakit.app:camera_imageJpegQuality) not found.
      
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: AAPT: error: attribute camera_imageMegaPixels (aka com.camerakit.app:camera_imageMegaPixels) not found.
      
  /Users/nabin/Desktop/camerakit-android/app/src/main/res/layout/activity_main.xml:29: AAPT: error: attribute camera_permissions (aka com.camerakit.app:camera_permissions) not found.
      
  error: failed linking file resources.
  Command: /Users/nabin/.gradle/caches/transforms-1/files-1.1/aapt2-3.2.1-4818971-osx.jar/0956a280f3a16fb98d7379b9aee64c63/aapt2-3.2.1-4818971-osx/aapt2 link -I\
          /Users/nabin/Library/Android/sdk/platforms/android-28/android.jar\
          --manifest\
          /Users/nabin/Desktop/camerakit-android/app/build/intermediates/merged_manifests/debug/processDebugManifest/merged/AndroidManifest.xml\
          -o\
          /Users/nabin/Desktop/camerakit-android/app/build/intermediates/processed_res/debug/processDebugResources/out/resources-debug.ap_\
          -R\
          @/Users/nabin/Desktop/camerakit-android/app/build/intermediates/incremental/processDebugResources/resources-list-for-resources-debug.ap_.txt\
          --auto-add-overlay\
          --java\
          /Users/nabin/Desktop/camerakit-android/app/build/generated/not_namespaced_r_class_sources/debug/processDebugResources/r\
          --custom-package\
          com.camerakit.app\
          -0\
          apk\
          --output-text-symbols\
          /Users/nabin/Desktop/camerakit-android/app/build/intermediates/symbols/debug/R.txt\
          --no-version-vectors
  Daemon:  AAPT2 aapt2-3.2.1-4818971-osx Daemon #0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 15s
21 actionable tasks: 1 executed, 20 up-to-date

```

wjhtinger, 2019-12-04:
I have the same problem, what can I do?

sliontc, 2020-06-01:
> I have the same problem, what can I do?