CameraKit/camerakit-android #590: E/libc: Access denied finding property "persist.camera.legacy_perf" Error

abesqaured, 2019-11-26:
### Environment
(Please include the following information along with any other relevant environment details.)

CameraKit Version: 1.0.0 version 3.11

Android Device: Moto Z Play

Android Version: 8.0.0

### Steps to Reproduce

(Write your steps here:)

1. I am building a simple camera app but I am unable to take photos or save them. All I have the camera view showing. I keep getting this error along with it unable to take or save pictures. 

2. I found this error that keeps coming up. 
E/libc: Access denied finding property "persist.camera.legacy_perf"

3. Here is my current code. 
  private CameraKitView cameraKitView;
    private Button button;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        cameraKitView = findViewById(R.id.camera);
        button = (Button) findViewById(R.id.button);
       button.setOnClickListener(photoOnClickListener);
    }

    @Override
    protected void onStart() {
        super.onStart();
        cameraKitView.onStart();
    }

    @Override
    protected void onResume() {
        super.onResume();
        cameraKitView.onResume();
    }

    @Override
    protected void onPause() {
        cameraKitView.onPause();
        super.onPause();
    }

    @Override
    protected void onStop() {
        cameraKitView.onStop();
        super.onStop();
    }

    @Override
    public void onRequestPermissionsResult(int requestCode, String[] permissions, int[] grantResults) {
        super.onRequestPermissionsResult(requestCode, permissions, grantResults);
        cameraKitView.onRequestPermissionsResult(requestCode, permissions, grantResults);
    }

    // From button OnClickListener
    private View.OnClickListener photoOnClickListener = new View.OnClickListener() {
        @Override
        public void onClick(View v) {
            cameraKitView.captureImage(new CameraKitView.ImageCallback() {
                @Override
                public void onImage(CameraKitView cameraKitView, final byte[] capturedImage) {
                    File savedPhoto = new File(Environment.getExternalStorageDirectory(), "photo.jpg");
                    try {
                        FileOutputStream outputStream = new FileOutputStream(savedPhoto.getPath());
                        outputStream.write(capturedImage);
                        outputStream.close();
                    } catch (java.io.IOException e) {
                        e.printStackTrace();
                    }
                }
            });
        }
    };

### Expected Behavior
It should be able to take photos and save photo but It is unable to do that.

XPHuster, 2020-05-06:
same question, waiting to solve

vaibhav936, 2022-01-10:
Same Question

vaibhav936, 2022-01-10:
2022-01-10 18:07:57.235 8439-8439/com.example.facesaveanddetetct E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2022-01-10 18:07:57.236 8439-8439/com.example.facesaveanddetetct E/libc: Access denied finding property "vendor.camera.aux.packagelist2"
2022-01-10 18:07:57.237 8439-8439/com.example.facesaveanddetetct E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2022-01-10 18:07:57.237 8439-8439/com.example.facesaveanddetetct E/libc: Access denied finding property "vendor.camera.aux.packagelist2"
2022-01-10 18:07:57.474 8439-8563/com.example.facesaveanddetetct E/libc: Access denied finding property "persist.camera.legacy_perf"
2022-01-10 18:08:04.708 8439-8439/com.example.facesaveanddetetct E/RequestQueue: cancel failed: no repeating request exists.