CameraKit/camerakit-android #593: Can't open documentation for v0.13.4

peterfortuin, 2019-12-12:
### Is this a bug report?

yes

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

yes

### Environment

CameraKit Version: 0.13.4

Android Device: not applicable

Android Version: not applicable

### Steps to Reproduce

1. Goto documentation page (https://camerakit.io/docs)
2. Select v0.13.4 on the top left corner

### Expected Behavior

Page should display documentation for v0.13.4 and show that versionnumber on the top left in the selection box.

### Actual Behavior

Page still shows documentation for v1.0.0-beta3.11

### Reproducible Demo

See above.

huangziqiangj, 2019-12-16:
I am also troubled by this problem

Silbernitrat, 2019-12-28:
I am also troubled by this problem

peterfortuin, 2019-12-28:
This project seems to be inactive. I suggest to look at this project:  https://github.com/natario1/CameraView